COVESA/vehicle_signal_specification #1: All rights reserved?

devcurmudgeon, 2016-03-06:
Hi,
It's great that JLR is leading this work, but would you consider downgrading the copyright claim?

Alternatively, if JLR intends to assert copyright, maybe guidelines for third-party contributions need to be stated?

magnusfeuer, 2016-03-07:
JLR would get copyright by default since I wrote this on the clock. However, the CC 4.0 allows for pretty much unlimited derivative work to be carried out.

I will double check with Matt if we can loosen the copyright claim, and will also define a contribution process.

closed 2016-12-12 by rstreif

gunnarx, 2016-12-22:
Disclaimer IANAL for all of the below:

Agreed - and basically all jurisdictions where Copyright exists, the creator has it by default according to the Berne convention, for a creative piece of work.  No assertion necessary... and without having the copyright, granting a license would not mean anything either.  I do not know what downgrading a copyright claim means, but perhaps you mean to grant a different license?

That said, I would call it uncommon to see "All rights reserved" written in combination with an open-source license, even if I do see it once in a while.  Writing it might be contradictory to the rights that *are* granted by the FOSS license.  I therefore second the request to remove those words for clarity.  (Do not remove the (C) which is necessary for adopters to know).

+1 for guidelines for third-party contributions.  As long as they are absent, contributors ought to specify their claim to copyright (they have it anyway, but for clarity) for all their modifications.  And they ought to assign an open-source license to those changes (preferably the same license* as the rest of the project, and as project maintainer it would be smart to require that for anything to be merged).   Contributors are not assigning copyright over to anyone unless the project requires contributors to sign/agree to a CLA.

Note also:
* There is a LICENSE file included (MPLv2) but no equivalent for the document license (CC-BY-4.0).  If you agree with the following that will be a moot point:

\* Reading between the lines it seems you want to license code and docs differently.  But it's not obvious to me why the YAML defined specification is considered a document (CC-BY) while Franca interface definitions are considered source code (MPL)?   For simplicity I would urge you to set the same license for all different data formats in this repository.  The data sets herein are of the kind that we should expect everyone to want to freely transform and combine them in various ways.  That is much easier with a single license - no worries about license incompatibility or annotating the end result incorrectly.

Payaspx, 2021-08-09:
> Hi,
> It's great that JLR is leading this work, but would you consider downgrading the copyright claim?
> 
> Alternatively, if JLR intends to assert copyright, maybe guidelines for third-party contributions need to be stated?

Payaspx, 2021-08-09:
https://github.com/GENIVI/vehicle_signal_specification/issues/1#issue-138844175

gunnarx, 2021-09-28:
> It's great that JLR is leading this work, but would you consider downgrading the copyright claim?

This is a 4 year old ticket(!)   "All Rights Reserved" is not necessary, I agree.  It is only confusing in the presence of a FOSS license that is clearly here.  Those words are no longer used in any of the official vspec.  They seem to be found in the private branch example, but that should be removed.

> Alternatively, if JLR intends to assert copyright, 

Well copyright does not really get "asserted" - it is there by default.  It is rather the conditions of use that are important.  But that's a nitpick.

> maybe guidelines for third-party contributions need to be stated?

The license conditions for use and derivative work are in the [LICENSE](https://github.com/GENIVI/vehicle_signal_specification/blob/master/LICENSE) file.

Third-party contributions are according to most standard open-source practice.  In other words, you keep your copyright on the information that you have authored, and your change proposals are accepted only under the same open-source license as the rest of the project. 

That should conclude this question for now.