COVESA/vehicle_signal_specification #5: Should the signal states include uncertainty?

bashrc, 2016-03-07:
If you want to use any of these parameters to do automated reasoning - especially for multi vehicle situations - then there's going to be a need to specify ranges of uncertainty or probability density functions.

rstreif, 2016-12-12:
@bashrc This is probably something one would do on a server backend where signals from many vehicles get consolidated. The specification can of course be extended to support fields for that however that has not been on the feature list. If you like please make a suggestion on how you would want to support it and for what signals it would make sense.

gunnarx, 2016-12-22:
Yes, I'd vote to keep the data herein clean and minimal.  It's really hard to anticipate everyone's needs and I would fear the YAML will end up being bloated with info and still not be appropriate for everyone.

Don't overlook that Franca IDL has got this stuff right.  The interface IDL is interface only!  Particular deployment details go into a separate deployment specification.  That's a place where the interface can be embellished with any information you choose.  So IMHO don't put such special things in the YAML but beside it.  ( But don't do that either because you're reinventing Franca :-) )

Regardless where you put it, a concrete example, yes.

rstreif, 2016-12-22:
One can easily make such extensions by defining signal overlays in separate, potentially private, files. This is an example how this would work for adding CAN fields to the signals:

Default definition for speed in Signal.Drivetrain.Transmission.vspec
- Speed:
  type: Int32
  min: -250
  max: 250
  unit: m/s
  description: Vehicle speed, as sensed by the gearbox.

If you now want to add a CAN field to it, you can do so in a private branch by Private/MySignals/CAN.vspec as:
- Drivetrain.Transmission.Speed:
  can: <candb-canframe>

(That's just an example. You can add any field you want and modify any existing field but not the type.)

Now you need to include the file with your toplevel vspec file:
#include Private/MySignals/CAN.vspec Signal

closed 2016-12-22 by rstreif