COVESA/vehicle_signal_specification #9: Makefile don't execute on OSX

peterMelco, 2016-06-21:
If after cloning rep and running make on Osx
you get:
**make: *** No rule to make target `vst', needed by `json'.  Stop.**

The current make file could be replaced with the following on OSx:
**# W3C vehicle data conversions
.PHONY: VERSION

VERSION=$$(cat VERSION)

all:
        ./tools/vspec2json.py -I ./spec ./spec/Vehicle/Vehicle.vspec > vss_w3c_$(VERSION).json**

rstreif, 2016-12-12:
@peterMelco Unfortunately I do not have access to an OSX system to test on. However, please make your suggested changes and submit a pull request. When submitting a pull request to resolve an issue on another platform please make sure that you also test it on Linux to be sure it did not break anything.

PeterWinzell, 2016-12-13:
I will get back on this as soon as possible - been a while since I looked at this.

as2902b, 2017-03-08:
I ll give this a shot tonight.

iamsanjeev, 2017-03-20:
I have tried this on the latest OS X version Sierra 10.12.3.

git clone <vss repo>
sudo pip install pyyaml

make works without any issues.

Here's the log I see.

vehicle_signal_specification sanjeev$ make
rm -f vss_rel_$(cat VERSION).json vss_rel_$(cat VERSION).fidl vss_rel_$(cat VERSION).vsi vss_rel_$(cat VERSION).csv
./tools/vspec2json.py -i:spec/VehicleSignalSpecification.id:1 -I ./spec ./spec/VehicleSignalSpecification.vspec vss_rel_$(cat VERSION).json
./tools/vspec2franca.py -v $(cat VERSION) -i:spec/VehicleSignalSpecification.id:1 -I ./spec ./spec/VehicleSignalSpecification.vspec vss_rel_$(cat VERSION).fidl
./tools/vspec2vsi.py -v $(cat VERSION) -i:spec/VehicleSignalSpecification.id:1 -I ./spec ./spec/VehicleSignalSpecification.vspec vss_rel_$(cat VERSION).vsi
./tools/vspec2csv.py -i:spec/VehicleSignalSpecification.id:1 -I ./spec ./spec/VehicleSignalSpecification.vspec vss_rel_$(cat VERSION).csv

I think we can close this issue. Someone likely fixed it.

rstreif, 2017-03-20:
@iamsanjeev Thanks for testing. Your output looks good. Let's wait for @PeterWinzell to confirm that it is working for him too.

peterMelco, 2017-12-01:
This is since long overdue. Closing this,

closed 2017-12-01 by peterMelco