COVESA/vehicle_signal_specification #22: Set a single license on the whole project

gunnarx, 2017-01-10:
I commented on a previous issue (#1) but it was already closed.  I would like to highlight what the last paragraph mentioned and request:

1. Set a single license for the whole project.
2. Clarify (e.g. in README) that contributions are to be made under that same license.
3. Clarify how contributors can make sure to certify their contribution's ownership and license.  The normal convention is by the contributor keeping the copyright themselves and licensing it under the same license and clarifying this by referencing [Developer Certificate of Origin](http://developercertificate.org/) -- [alt link 1 **with explanation**](http://elinux.org/Developer_Certificate_Of_Origin) -- [alt link 2](http://www.eclipse.org/legal/DCO.php) on the first contribution.  Thereafter a sign-off line on each commit.

Rationale:

Reading between the lines it seems you wanted to license code and docs differently. But it's not obvious to me why the YAML defined specification is considered a document (CC-BY) while Franca interface definitions are considered source code (MPL)?  The data in this project are of the kind that we should expect everyone to want to freely transform and combine them in various ways. That is much easier with a single license - no worries about license incompatibility or annotating the end result incorrectly.

gunnarx, 2017-01-10:
Other notes.
Currently the project has a LICENSE file included (MPLv2) but no equivalent for the document license (CC-BY-4.0). If you agree with using one license only, that will be a moot point.

gunnarx, 2017-01-10:
Example case:

See the original (CC-BY licensed) [Body.vspec](https://github.com/GENIVI/vehicle_signal_specification/blob/a53a82d214c7de77b14a773ea783cbf555dd160c/spec/Attribute/Body.vspec) vs. a file like this which would presumably be generated from vspec (see the RefuelPosition signal) [vss_rel_1.fidl](https://github.com/GENIVI/vehicle_signal_specification/blob/a53a82d214c7de77b14a773ea783cbf555dd160c/vss_rel_1.fidl)  (MPL licensed)

rstreif, 2017-01-11:
All files and artifacts have now been unified to MPLv2. See pull request #23.

closed 2017-01-11 by rstreif