COVESA/vehicle_signal_specification #27: Temperature Domain unit

ssivasan, 2017-02-24:
Can you please confirm my concern, For temperature the unit is mentioned as Celsius, say example, if UI display expects the temperature to be displayed in Fahrenheit, but the respective vehicle data is in Celsius. The conversion can happen in native code before transmitting to browser (or) in the browser itself.

Is there any W3C standard such that always the temperature unit is realized as Celsius.Means the data transfer from native software to browser should always happen in Celsius?

rstreif, 2017-02-24:
@ssivasan Yes, you are correct, conversion can happen where it is appropriate for your application.

I am not sure if there is a W3C standard requiring temperature measurements in degree Celsius.

ssivasan, 2017-02-27:
Thanks for the update.

closed 2017-02-27 by ssivasan