COVESA/vehicle_signal_specification #28: README.md

ssivasan, 2017-02-24:
Few spelling to be corrected in the below topics.

ATTRIBUTE ENTRY - the "specifiation", that never changes

SIGNAL NAMING CONVENTION - the "reaview" mirror

ATTRIBUTES - data that can be specified when the "bspec" file is installed.

EXTENDING AND OVERRIDING SIGNALS - Fig 4. "Overidden" signals
                                                                     vss_23.vspec and then "oveerriding" the

The Quoted item needs correction.

rstreif, 2017-02-24:
@ssivasan 

Thank you for submitting these corrections. I am not sure if you are familiar with the GitHub contribution process. I would like to ask you to simply make the corrections and submit a pull request for me to merge. Thank you.

ssivasan, 2017-02-27:
I have submitted a pull request with commit ID - 73a9ad8. Thanks.

closed 2017-02-27 by ssivasan