COVESA/vehicle_signal_specification #41: Need to support fraction of inches?

erikbosch, 2017-08-14:
Is there a need to support fraction of inches? At least for e.g. buses it is not uncommon with wheel sizes like 22.5" x 8.25".  If needed one could consider a representation like "milli-Inches", e.g. "22500" for a 22.5" wheel

Is the standardisation organisation not called ETRTO (ETRO in current spec)
http://www.etrto.org/Home

https://github.com/GENIVI/vehicle_signal_specification/blob/222bbc6d1900e2c82dc15138da2266dcdae3c53e/vss_rel_1.0.fidl#L139

gunnarx, 2018-01-29:
Note that the FIDL file is auto-generated and just a test of a translation. Actually it's not perhaps the right way to do such a translation, in my opinion, so it might need to be looked at).  In any case, the "official" spec are the YAML files - please point to the right place there instead.

gunnarx, 2018-01-29:
For obviously wrong ETRO, please provide pull request.

rstreif, 2018-03-26:
VSS should use SI units in genera. Only exception would be where the automotive industry has established standards using different units.
VSS supports the `float` data type for fractions.
We need to go through VSS and check for consistency.

commit 0ed437d1 (2021-06-08): Updating specification for wheels and tires

Fixes #41

Updating specification to work better with real examples

Rationale:
- Rim diameter and width are typically given as fraction of inches,
  like 8.25 x 22.5 as an alternative for a Scania truck
- TireWidth is normally given as millimeters,
  with 295/80R22.5 as an example for the same Scania truck
- If rim size and tire width is mentioned, then mentioning aspect ratio
  as well makes sense, as those are the 3 most important data if you buy
  a new tire