COVESA/vehicle_signal_specification #42: Is support for TireWidth in inches required?

erikbosch, 2017-08-14:
Most cars, at least in Europe, use millimeters to represent tirewidth, e.g. 195/60-R15, typically with 10 mm increments, e.g. 185,195, 205, ... So is Inch as unit requested by OEMs, especially if it does not support fractions of inches?

https://github.com/GENIVI/vehicle_signal_specification/blob/222bbc6d1900e2c82dc15138da2266dcdae3c53e/vss_rel_1.0.fidl#L163

gunnarx, 2018-02-19:
Comment says "per ETRO / TRA standard" (which should be [ETRTO](https://www.etrto.org/Home) as noted by another. )

I don't know who is enough of a tire expert to answer, but at least a clear reason is given, so I'd say that's OK (assuming inches is actually the ETRTO standard, which seems surprising as it is a European organization?).  The policy (README) ought however to clarify if all signals are to be SI units, or if exceptions are allowed when there is a good reason for the exception.

Whatever the right answer is, agree it must surely be including fractions of inches, unless the unit is as small as millimetres, if this is to be used for calculating distances etc.

rstreif, 2018-03-26:
Falls also under #41.

closed 2018-03-26 by rstreif