COVESA/vehicle_signal_specification #44: Altitude with double precision?

erikbosch, 2017-08-14:
In general the selection of datatype is not always easy to understand, for example temperature (celsius) can be found with int8, int16 and float. I know there might be historical reasons in some cases, or that datatypes form other standards have been used, but that is difficult to know as a reviewer. I assume that altitude is defined as ground level

Maybe there have been

https://github.com/GENIVI/vehicle_signal_specification/blob/222bbc6d1900e2c82dc15138da2266dcdae3c53e/vss_rel_1.0.fidl#L2859

gunnarx, 2018-01-29:
Sometimes I think differing ranges, as well as precision, might be considered when selecting datatypes even for the same physical unit - but temperature range cannot vary *that* much between different items.  (It is not intended to be applied to the [LHC](https://www.seeker.com/lhc-smashes-highest-man-made-temperature-record-1765929082.html)  :-)

For CAN signals and the like, there are different encodings, but that extends not only to unit and number of bits, but also to the concept of scaling, which is usually a first order equation (but theoretically I suppose higher order could be applied).   But for the purpose of a shared API on some level, which is the only thing this specification can aim for here, I'd agree that a single datatype for all data of the same unit ought to be applied unless there are good exceptions.  Certainly for temperature, I agree.  Scaling is not really applied here what I can tell, so higher precision than 1 degree suggests it should be a float.  Why is it double?  Have no idea...

A comment could specify the intended definition of Altitude (Just make pull request) even if it is likely assumed by everyone that it is an absolute term and that there can be no other reasonable reference than sea level.  Noteworthy, [according to Wikipedia](https://en.wikipedia.org/wiki/Altitude) the preferred term to use, to be really sure that sea level is the reference, is "Elevation".

rstreif, 2018-03-26:
Duplicate of #41. If there are inconsistencies in the use of the data type for the same unit we need to address this.

closed 2018-03-26 by rstreif