COVESA/vehicle_signal_specification #45: -        Navigation Destination:  should other options be available besides Long, Lat.    Like Address, UK postcode etc.

Geraldbosch, 2017-08-15:


gunnarx, 2018-01-29:
Since this is a "signal" API as opposed to a full Navigation function API, I'd personally think some more advanced options are out of scope.   It is better discussed in a Navigation API setting.  

But that said, it's not fully clear which level this Infotainment part of the spec is aiming for, so I would propose a community discussion on that.   I suppose it is aimed at information that Infotainment needs to communicate (as signals) to other ECUs, as well as ways to control Infotainment from other ECUs.  That is an area that might be hard to find common ground on, but whatever it is, it definitely ought to sync'ed with other definitions, starting with for example GENIVI API definitions for Media and Navigation, etc.

rstreif, 2018-03-26:
Originally VSS was meant just for signals. However, latest developments among them in the context of the W3C Automotive Working Group is looking at extensions to cover location based services, multi media, etc. Proposal here is to wait for the results of the collaboration.

danielwilms, 2021-06-01:
Please reopen, if it's an issue still.

closed 2021-06-01 by danielwilms