COVESA/vehicle_signal_specification #46: Transmission Drive Type  enum:  abbreviation not really clear “fwd”, “rwd”

Geraldbosch, 2017-08-15:


gunnarx, 2018-01-29:
File, and line, please

gunnarx, 2018-01-29:
I assume it's fwd = forward wheel drive, rwd = rear whell drive, awd = all wheel drive.
Should there be a comment to clarify?  Please suggest a change through PR.

klotzbenjamin, 2018-12-11:
In https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Drivetrain/Transmission.vspec, it has been corected already:

`- DriveType:`
`  datatype: string`
`  type: attribute`
`  enum: [ "unknown", "forward wheel drive", "rear wheel drive", "all wheel drive" ]`
`  value: "unknown"`
`  description: Drive type.`

@danielwilms we can close this issue.

danielwilms, 2018-12-11:
Hi @klotzbenjamin, great. There's one open question for me: Do we have a common way to define enums? The spaces look weird.

klotzbenjamin, 2018-12-11:
> Do we have a common way to define enums? The spaces look weird.

I do not see any rule for defining enums, in this case the only rule is that the elements from the enum are of `datatype: string`.

danielwilms, 2018-12-11:
OK, I'll close this one then and open a new one for uniform enum definition.

closed 2018-12-11 by danielwilms