COVESA/vehicle_signal_specification #47: The mirror  heater status seems to be boolean,  whereas the windscreen heater seems to be a string/enum

Geraldbosch, 2017-08-15:


gunnarx, 2018-01-29:
Generally:  Please try to provide exact file/line reference, (or even better of course a change request / pull request if you know what the change should be).  Otherwise it is redundant work for other people to search for what you mean.  (For example what you call "windscreen" seems to be called "WindShield" in the file -- at least now it is.  Such things make it harder)

You can do this by opening the file in GitHub and clicking on a line number.  The URL is then in the Browser bar, or by clicking the "..." menu that appears and selecting "Copy permalink"

A link looks like this, for line 67 of Body.vspec:

```
https://github.com/GENIVI/vehicle_signal_specification/blob/222bbc6d1900e2c82dc15138da2266dcdae3c53e/spec/Signal/Body/Body.vspec#L67
```
which GitHub then displays nicely like this:
https://github.com/GENIVI/vehicle_signal_specification/blob/222bbc6d1900e2c82dc15138da2266dcdae3c53e/spec/Signal/Body/Body.vspec#L67

gunnarx, 2018-01-29:
For this one, it seems like something was fixed there already 5 of May 2017 in commit 198a022 by @iamsanjeev?  Are there still more like that?
@Geraldbosch Do you know which branch were you looking at when this was sent (15 August?)

chheis, 2021-01-22:
@gunnarx I would say, it is fixed. (Release 2.0). Close it?

danielwilms, 2021-06-01:
fixed, and can be closed

closed 2021-06-01 by danielwilms