COVESA/vehicle_signal_specification #48: While issuing a  set  using a wildcard it seems that you still need to list all options in the value field. Is this the only option?

Geraldbosch, 2017-08-15:
" client -> {
               "action": "set",
               "path": "Signal.Cabin.Door.*.IsLocked",
               "value": [ {"Row1.Right.IsLocked": true },
                               {"Row1.Left.IsLocked": true },
                               {"Row2.Right.IsLocked": true },
                               {"Row2.Left.IsLocked": true } ],
               "requestId": "5689"
        }
o   This could be quite a lot for example  Seat. Row.Pos.heating would be about 20 entries. Some of these might even not exist and would need then additional error handling.

Maybe something like this

client -> {
               "action": "set",
               "path": "Signal.Cabin.Door",
               "value": [ {"*.*.IsLocked": true },
                        ],

gunnarx, 2019-04-16:
I'm not sure I've heard all possible proposals for queries.  But indeed the first one (current spec? if I understand @Geraldbosch?) indeed makes less sense to me.  I think the usefulness of a wildcard/query in path disappears if each item needs to be mentioned explicitly anyway.

The second proposal doesn't quite feel right either:
```
 "path": "Signal.Cabin.Door",
"value": [ {"..IsLocked": true },
```

What does "..IsLocked" imply?  Is there an implicit wildcard between path and value here, and does it match any number of levels of tree?  So that:
**Signal.Cabin.Door.one.two.three.four.IsLocked** would be set to **true** ?

Generally speaking, why not have a fairly simple wildcard/query applied on path, which is pattern-matched against the full paths of the actual objects, and then provide a single plain value that will be assigned to all matched objects.  To set different values you would have to send different Set commands.

The original example would then be:

client -> {
 "action" : "set",
 "path": "Signal.Cabin.Door.*.*.IsLocked
 "value": "true"
}

(This is of course more or less the original wildcard matching semantics, isn't it?)

If those are the query semantics, it does require the client to have at least the knowledge that there are two levels of hierarchy (Row and Left/Right) between Door. and .IsLocked but that seems reasonable to me so that you don't start matching totally unexpected things.

I know the proposal #81 would shift the order of the tree around, but the simple wildcard matching would work also after such changes?

gunnarx, 2019-04-16:
@danielwilms Perhaps this question is rather obsolete now because of discussion had in #81 and ought to be closed? 

 Or it needs to be updated since #81 clarifies tree structure and how to encode positional parameters, but it might not be documented (I think) how a Set operation shall function on a path query on to that new tree structure?  (Is there a possibility/requirement to give multiple/different values for matching objects or only one?)

danielwilms, 2021-06-01:
should be fixed through #81, if not please reopen.

closed 2021-06-01 by danielwilms