COVESA/vehicle_signal_specification #52: Signal ID inconsistencies between different formats

SebastianSchildt, 2018-02-16:
Hi, I was under the impression, that the different formats should contain the same info, however at least for .csv and .json that does not seem to be true

```
$ cat vss_rel_1.0.json | grep -A 5 ConsumptionSinceStart
              "ConsumptionSinceStart": {
                "description": "Fuel amount consumed since start in milliliters.", 
                "type": "UInt32", 
                "id": 82, 
                "unit": "ml"
              }, 

$ cat vss_rel_1.0.csv | grep ConsumptionSinceStart
Signal.Drivetrain.FuelSystem.ConsumptionSinceStart,"81","UInt32","ml","","","Fuel amount consumed since start in milliliters.",""
```

One is id82, but the other is id81. Is this intended (why?), or a bug?

klotzbenjamin, 2018-02-16:
I just checked the repo and when you generate again the spec, the now have the right IDs.

gunnarx, 2018-03-26:
Not sure I understood but isn't it simply a (bad) consequence of adding generated files (JSON, CSV as well as .id) to the repository, and that those files happen to be out of sync right now?

On the one hand, it's bad practice - normally there should be **one** source of information only (the YAML files are the official files), and anything that can be (re-)generated should not be version controlled.  

On the other hand, including CSV and JSON formats help casual users to see what they look like and that they are supported, without having to run the tools.  It's a trade-off and because of the early stages of standardization and usage here, I would personally update the generated files for a while, but make a special note that they are not the official files, and could be outdated.   (In the longer run, the generated files should not be part of the repository however IMHO)

If I misunderstood the issue or there is more discussion on how IDs ought to be generated, let's hear it.  

(Note: I am not the maintainer but @rstreif is)

rstreif, 2018-03-26:
This is related how the ID file is generated and used. When the format converters are run the core checks if the ID file is already present. When it is then it is used and the IDs are not changed. However, if the name or branch of a signal changes it will create a new entry in the ID file. Deleting the ID file fixes this issue.
The principal idea of the ID file is to remain somewhat backwards compatible but it only makes sense once signal names and branches have settled.

danielwilms, 2020-07-15:
Implementation changed of the ID file generation has changed completely. Issue resolved now, as hashes are created in a different way.

closed 2020-07-15 by danielwilms