COVESA/vehicle_signal_specification #56: OBD2 ?

SecureCloud-biz, 2018-03-08:
Can this be used with an OBD2 plug to pull info from a vehicle?

klotzbenjamin, 2018-03-08:
Hi,
I am not sure what you mean.

VSS contains an [OBD branch spec](https://github.com/GENIVI/vehicle_signal_specification/blob/develop/spec/Signal/OBD/OBD.vspec) in which are described a certain number of OBDII signals (from Pid 00 to 5E).

You can use it to structure  and annotate data you would extract using an OBD plug, for instance with the type, unit and description of a signal.

gunnarx, 2018-03-26:
@SecureCloud-biz: The vehicle signal databases are not yet common enough between car companies.  (Often very little beyond what is legally required, is even published by car OEMs, so most third-party OBD-2 tools rely on reverse engineering).

This project aims to find some common ground on defining vehicle signal data.  If sharing the definition of the low-level signals fails,  the project *still* defines a shared way to describe signal databases, shared tooling for working with it, and it all connects into a larger ecosystem of technologies, such as [W3C standards](http://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html)

The database stored here is on a bit more abstract level and not really tied to a specific bus (such as CAN). The format is however intended to **enable** inclusion of additional information, such as exact signal IDs using industry standard "CANdb" format.   

So... the short answer is: No, because the information you need (exact CAN signal frame IDs,  encoding, etc.) is not in this public database.

closed 2018-04-17 by SecureCloud-biz