COVESA/vehicle_signal_specification #79: [vss2] Attribute and signal as type

danielwilms, 2018-11-27:
**Current situation:**
Right now signals and attributes are modeled in the tree structure itself. Types can be either branch or data types. Some issues resolve from that modeling choice:
* redundancy of tree structure in both branches
* logically combined leafs are separated
* graph type and data type mixed
* private branch doesn't effect attribute tree
* property in graph structure hard to change

Further there is #78 proposing a solution and #70 raising the issue.

**Possible solution**
1. Taken from #78 Attribute and Signal branch would be removed
1. Type would be changed from ['Branch', DataType1, ...  n] to ['Branch','Signal','Attribute']
1. New element datatype would be introduced as element.

@UlfBj, could you check, if I forgot something?

UlfBj, 2018-11-27:
Thank you for asking about my opinion.

I think you have got the arguments right for the current situation.

In the Possible solution below I think that the name ‘Signal’ is problematic. ‘Signal’ represents both the ‘Sensor’ and ‘Actuator’ types, as well as their combination ‘SenseActuator’ (my naming).
I have also suggested that we should represent “streaming” data sources with a specific type ‘StreamSensor’.
And that diagnostic type of data should have its own type, ‘Diagnostic’.
With this said, I think that Type should be represented the set ['Branch', 'Sensor', 'Actuator', 'SenseActuator', 'StreamSensor', 'Diagnostic', 'Attribute'].  The naming may be changed, but this is the set of types I think are relevant.

Further, if my contribution of the node types ‘Rbranch’ and ‘Element’ also should be taken into account, I guess they should also be added to the Type set.

I think it is a good idea to separate between Type and Datatype.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

  2.  Type would be changed from ['Branch', DataType1, ... n] to ['Branch','Signal','Attribute']
  3.  New element datatype would be introduced as element.

gunnarx, 2018-11-27:
I'd like to (help out to) get #78 merged as soon as possible so that things don't get out of date again from git point of view.  #78 is still in conflict status and it requires a kind of special merge because of this -- based on Ulf's input we aim to essentially overwrite master with the content of #78.  But that should be done technically using a set of git-merge steps if we  want the history to include master rather "resetting" master to new commit, which is usually not done for commit histories that have been published).  I have done this locally but want the group's input on whether this can be pushed onto master.  Once it is done the future work becomes easier to do (from a pure git technical point of view).

danielwilms, 2018-11-28:
@gunnarx: From a git perspective you're completely right. The change is huge and should be merged ASAP. Some things I still would like to consider to change. 

I summarize from the discussion here:
1. attribute signal branch will be removed
1. datatype will be a new element with datatypes only
1. type will be ['Branch', 'Sensor', 'Actuator', 'SenseActuator', 'StreamSensor', 'Attribute']

One thing I'm not sure about is Diagnostic as a type, because at the end it's just a protocol which transmits the sensor value or acts on the actuator. Another is the term SenseActuator. I don't have a better suggestion right now, but it sounds a bit odd to mix a verb and a noun. Suggestions?

@UlfBj how shall we do it practically? Do you have today to work on this? Would you need support? Tomorrow night we have a call on the v2 changes, it would be cool to have the PR until then mergeable. I can help if needed.

UlfBj, 2018-11-28:
Hi,

The proposed changes in bullets 1 to 3 is fine with me.

Regarding the Diagnostic type, my view is that e. g. DTC data does not so well fit into any of the available Type values in bullet 3. A VSS node SHALL have a Type, so which one in bullet 3 should be used for DTC tree nodes?
But I do not have a strong opinion on this, it is fine to omit it.

I agree that SenseActuator is a somewhat odd name. I tried hard to find a better name, but I failed. However, I think this type needs a name of its own.

Regarding how we should do this practically, I am not the best to answer that, my Github knowledge is rather shallow. So I need help to do this. If Gunnar or someone else is willing to take the lead, I think that would be the best way forward. I can then be available to answer any questions regarding what I have tried to do in the pull request.

Please let me know how you think we should go about this.

It would be cool to have it ready for the meeting you mention☺.

BR
Ulf

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

  2.  datatype will be a new element with datatypes only
  3.  type will be ['Branch', 'Sensor', 'Actuator', 'SenseActuator', 'StreamSensor', 'Attribute']

UlfBj, 2018-11-28:
>> A DTC can be seen as an error on a specific functionality and maybe should be modeled there. Maybe we need some examples to see how we model this one.
Ok.

>> I can take it from there. I'll see how much has to be changed the worst case, I'll just create a new one.
Great. If any questions emerge that you think I may have an answer on, I am available most of the day via mail, or other means.

BR
Ulf

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

A DTC can be seen as an error on a specific functionality and maybe should be modeled there. Maybe we need some examples to see how we model this one.

I can take it from there. I'll see how much has to be changed the worst case, I'll just create a new one.

gunnarx, 2018-11-28:
>> A DTC can be seen as an error on a specific functionality and maybe should be modeled there. Maybe we need some examples to see how we model this one.

>Ok.

I agree, but there will need to be a lot of examples, or it is likely to be modeled in a way that is not convenient.  Some DTCs might be standard (e.g. described in a ISO diagnostics or AUTOSAR spec), but most of them are OEM specific so that is worth keeping in mind.  

Besides, diagnostics involves a number of other aspects so I would suggest to not model this without consulting diagnostic specialists.  There are more concepts such as freeze-frames and fault indications (signals).  Before a DTC is set there may first be fault indications, and individual rules for each one.

Random example:  This fault indication must exist for 7 seconds and _then_ this DTC is set.  If the fault signal later disappears for 1 hour, this DTC is cleared.  But for another one the rule is that the DTC is never cleared.  You get the idea.  Now I realize that those rules are unlikely to be encoded in this database per se, I'm just saying the full picture should be understood before designing it.
 
A strategy could of course be started in terms of "ontology" how a DTC should relate to its function and to its signals.  But that would be working on a general strategy, probably not yet ready to be put into a specific database until all details are known.

If we expect most DTCs to be OEM specific, then keeping them in a separate tree could make it a bit easier and it would "postpone" the more advanced design for a  while.

gunnarx, 2018-11-28:
> @gunnarx: From a git perspective you're completely right. The change is huge and should be merged ASAP. 
> Some things I still would like to consider to change.

Please advise more clearly  - shall I make the merge to master now, or wait until that change discussion is done?

danielwilms, 2018-11-28:
> Shall I make the merge to master now, or wait until that change discussion is done?

Wait @gunnarx , I'll add some changes to the PR before.

rtroncy, 2018-11-28:
>     3. type will be ['Branch', 'Sensor', 'Actuator', 'SenseActuator', 'StreamSensor', 'Attribute']

Regarding the list of possible types, as others have noted, I also agree that 'SenseActuator' is bizarre as naming. More importantly, why do you need it in a first place? To act the fact that some sensors may be both sensors and actuators? Multy-typing is fine, and generally well-implemented in schemadata and in particular ontologies. For examples, schema.org makes heavy use of Multi Typed Entities (MTE is a modeling pattern).

UlfBj, 2018-11-28:
So you propose it would be expressed like this?
‘Type’ = [ ‘Sensor’, ‘Actuator’ ]

Adds a bit to implementation complexity, but makes the specification cleaner.

BR
Ulf

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

UlfBj, 2018-11-28:
Or could we say that ‘Actuator’ in almost all cases is equal to [ ‘Sensor’, ‘Actuator’ ], and handle the (few?) exceptions when the client tries to read a non-readable ‘Actuator’ with an error code?

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

rtroncy, 2018-11-28:
> So you propose it would be expressed like this? ‘Type’ = [ ‘Sensor’, ‘Actuator’ ] Adds a bit to implementation complexity, but makes the specification cleaner. 

Yes, for some specific Type that are indeed both sensor and actuator.

danielwilms, 2018-11-28:
>>So you propose it would be expressed like this? ‘Type’ = [ ‘Sensor’, ‘Actuator’ ] Adds a bit to implementation complexity, but makes the specification cleaner.

>Yes, for some specific Type that are indeed both sensor and actuator.

Thanks for the input @rtroncy. The only problem I see is, that it adds complexity, as not every combination would be permitted (e.g. branch, attribute are mutually exclusive to the others). That makes me think as well, if StreamSensor is needed. Which update interval makes a sensor a streamsensor? To keep it simple, I would suggest:
_Branch_: node in the tree
_Sensor_: read-only which updates in some interval x
_Actuator_: Sensor + write
_Attribute_: read-only and static

What do you think?

UlfBj, 2018-11-28:
My view on streaming sensor is not related to what update interval it has, but that it uses a streaming protocol. Thus this API can be used for camera/radar data streams by providing the url to the streaming server. The actual stream is not handled by this API. We discussed this in the W3C WG where it was backed up.
BR
Ulf

Skickat från min Samsung Galaxy-smartphone.

rtroncy, 2018-11-28:
> The only problem I see is, that it adds complexity, as not every combination would be permitted (e.g. branch, attribute are mutually exclusive to the others). That makes me think as well, if [...]

This is a good start. And I would add _Stream_ as well for the case of @UlfBj mentioned. You may define your types as perfectly disjoint, as you seem to do when defining _Sensor_ as read only and _Actuator_ as sensors you can write. This means that there is no _Actuator_ that is not a _Sensor_ as well?
As for enabling some combination of types (or not), this is a different issue if you follow the multi-typing modeling pattern and this can be constrained with shapes (or rules). In summary, it seems to me we agree on the top / main types so let's push this in the repo

danielwilms, 2018-11-28:
> My view on streaming sensor is not related to what update interval it has, but that it uses a streaming protocol.

Sorry, makes sense. Then I misunderstood.

> In summary, it seems to me we agree on the top / main types so let's push this in the repo

Agreed. I will prepare it then.

danielwilms, 2018-12-03:
@zourongrong, thanks for the feedback. The discussion went a bit away of the SenseActuator, even though the context behind it hasn't changed that much. First, I'd say that CAN is not the only target bus we have, but we should definitely consider the consequences there. Please checkout the [previous comment above](https://github.com/GENIVI/vehicle_signal_specification/issues/79#issuecomment-442559481) and let me know what you think.

danielwilms, 2018-12-03:
@UlfBj I've done the changes and currently writing the documentation. There I came along the rbranch. Could you quickly help me out, what's the status with that one? This type should stay, am I right?

zourongrong, 2018-12-03:
> @zourongrong, thanks for the feedback. The discussion went a bit away of the SenseActuator, even though the context behind it hasn't changed that much. First, I'd say that CAN is not the only target [...]
@danielwilms Thanks for your reply. My question is: Is there any example about how to use these data models?
Maybe it shouldn't be discussed here :).

danielwilms, 2018-12-03:
@zourongrong, here's the perfect place to ask. I'm pushing the changes today and then we can discuss on the concrete documentation. I think that might be easier.

danielwilms, 2018-12-03:
> This has been thoroughly discussed in the W3C WG, where it was approved.
So I would say that it should stay, together with the ‘element’ type.
Naming could be discussed if a need is seen of that.

@UlfBj Thanks for the clarification. That's what I thought. Just wanted to be sure. :)

commit eb9408f2 (2018-12-05): [Prep2.0.0] Fix upper case datatypes

uniform way of defining datatypes

Fixes #79
closed 2018-12-05 by klotzbenjamin