COVESA/vehicle_signal_specification #84: Top nodes in VehicleSignalSpecification.vspec

UlfBj, 2018-12-07:
Under the root node Vehicle there is now, besides the Media node and the Private node, also also all the nodes like DriveTrain, Cabin, Body, ADAS, Chassis, and OBD. 
I think these should be pushed one level down, under a branch node that may have the name Car (from VIWI), or something else. The nodes mentioned above belong to the same logical domain, and should reside under a domain name node for these.  It will be more comparable to the Media domain node, and further new domain nodes that might be added later.

danielwilms, 2018-12-10:
> I think Chassis Body, Drivetrain, ADAS, etc should be placed together under one branch node, as I see them logically to belong to the same domain. But as it is hard to find a good name for that domain, maybe that indicates it is not a coherent logical group.
But this is more a matter of taste, I will not insist on my view.

If we find a domain within vehicle to group them, we can of course discuss it. I just would really like to avoid duplications in the path like, vehicle -> car -> ... I second your comment, that I don't see the logical connection right now either.

> What I would like to insist on, however, is that the inclusion shown below is changed.
#include Vehicle/Vehicle.vspec Vehicle
This clutters the level directly under the root with nodes like IgnitionOnTime, etc.
I think this should be gathered under a branch node, and not be directly under the root.
The Vehicle.vspec file name is not so well chosen either, being the same as the root name.

Do I understand right, that you want to have on the first level branches only? It would be a good compromise IMO. I'll think of a naming proposal. If you're faster, go ahead :)

UlfBj, 2018-12-10:
>> Do I understand right, that you want to have on the first level branches only?

Yes.

>> It would be a good compromise IMO. I'll think of a naming proposal. If you're faster, go ahead :)
Right now I do not have a proposal, but I will also think of it.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

UlfBj, 2018-12-10:
Here is my proposal:

In Vehicle.vspec the vehicle identification data is already under the branch VehicleIdentification.

The following set, under Vehicle signals, I propose that all, except AmbientTemperature and IsMoving, is put under a branch called Statistics.

The following sets Spacial acceleration and Spacial rotation, I propose is put under a branch called Inertia, where also isMoving goes.

The attributes at the end goes under the VehicleIdentification branch. The accelerationTime is questionable to me.

Where AmbientTemperature goes I do not know.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

rtroncy, 2018-12-10:
Some comments:
 * +1 for having a general root node named "Vehicle"
 * +1 for grouping Chassis Body, Drivetrain, ADAS and some more (what is the exhaustive list?) under a common branch name

> Here is my proposal: In Vehicle.vspec the vehicle identification data is already under the branch VehicleIdentification. The following set, under Vehicle signals, I propose that all, except AmbientTemperature and IsMoving, is put under a branch called Statistics.

Statistics? This seems a odd name for grouping vehicle signals. What did you mean exactly?

> The following sets Spacial acceleration and Spacial rotation, I propose is put under a branch called Inertia, where also isMoving goes.

+1

UlfBj, 2018-12-10:
>> Statistics? This seems a odd name for grouping vehicle signals. What did you mean exactly?
Most of them are accumulated values over time, IdleTime, DriveTime, etc.

I see now that one breaking that pattern is Speed, so it should be put elsewhere.
Statistics is maybe not the best name, ‘Accumulated time’ is probably better.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

danielwilms, 2018-12-10:
>> Statistics? This seems a odd name for grouping vehicle signals. What did you mean exactly?
Most of them are accumulated values over time, IdleTime, DriveTime, etc.

>I see now that one breaking that pattern is Speed, so it should be put elsewhere.
Statistics is maybe not the best name, ‘Accumulated time’ is probably better.

How about ```Vehicle.TripRecorder```?

UlfBj, 2018-12-10:
>> How about Vehicle.TripRecorder?
Better than Accumulated time.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

UlfBj, 2018-12-10:
Speed could go under Inertia (if we go with that).

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

rtroncy, 2018-12-10:
> How about `Vehicle.TripRecorder`? @rtroncy, @klotzbenjamin

You add a notion of `Trip` that does not really exist. When are those values being reset? When you switch on/off the engine? When the engine car is going in sleeping mode? `Recorder` is good as the underlying signals will indeed capture a phenomena that is accumulated over a period of time from my understanding. Again, what is the exhaustive list of signals that will be grouped under this branch?

UlfBj, 2018-12-10:
They are found in the Vehicle.vspec file.
https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Vehicle/Vehicle.vspec

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

danielwilms, 2018-12-10:
> You add a notion of `Trip` that does not really exist. When are those values being reset? When you switch on/off the engine? When the engine car is going in sleeping mode? `Recorder` is good as the [...]

I like the proposal to use simply ```Vehicle.Recorder```. Then I would put values like speed and ambientTemperature under it as well. First I thought about ```Vehicle.Status``` but that's maybe a bit too generic.

UlfBj, 2018-12-10:
>> I like the proposal to use simply Vehicle.Recorder.
+1

>> Then I would put values like speed and ambientTemperature under it as well.
Speed I think is more suitable in Inertia.

Ulf Björkengren Ph. D.
Connectivity Strategist

M +4553562142
ulf.bjorkengren@volvocars.com<mailto:ulf.bjorkengren@volvocars.com>

VOLVO CAR CORPORATION
94014 Lund R&D Tech Center
Frederikskaj 10A
Copenhagen, Denmark
volvocars.com

klotzbenjamin, 2018-12-10:
> I like the proposal to use simply Vehicle.Recorder. 

+1

danielwilms, 2018-12-10:
>> Then I would put values like speed and ambientTemperature under it as well.
> Speed I think is more suitable in Inertia.

@UlfBj: is there any other signal you'd see there, besides speed?

UlfBj, 2018-12-10:
Not in Vehicle.vspec.
I have not checked other files.

Skickat från min Samsung Galaxy-smartphone.

UlfBj, 2019-01-03:
@rtroncy :+1 for grouping Chassis Body, Drivetrain, ADAS and some more (what is the exhaustive list?) under a common branch name

After discussing internally about a possible  common branch name, the best we could come up with was "VehicleData". 

Our view was that even if it is not a clean homerun, it is an improvement compared to having all these branches directly under the Vehicle root node. 

Comments are welcome:).

danielwilms, 2021-06-01:
done.

closed 2021-06-01 by danielwilms