COVESA/vehicle_signal_specification #87: Command branch may be needed.

magnusfeuer, 2018-12-07:
JLR may have a need to define a set of command signals to modify the behavior of the vehicle.

One example would be `Signal.Cabin.Door.Row1.Right.Window.Position`, which reports the current window position. 

This signal would be paired with with `Signal.Cabin.Door.Row1.Right.Window.SetPosition`, which is picked up by the door/window controller to start rolling the window up and down. During the operation the `[...].Window.Position` would be continuously published with new values.

With this in mind I have a few questions for the VSS stakeholders:

1. Is this a good idea, or should we define commands via some other, non-VSS specification?

2. Assuming that the idea is not garbage, do we want to group all commands in their own top-level branch or closer to the leaf?<br>
`Command.Cabin.Door.Row1.Right.Window.SetPosition`
vs. 
`Signal.Cabin.Door.Row1.Right.Window.SetPosition`

3. Are there other things to take into consideration?

/Magnus F.

magnusfeuer, 2018-12-08:
We are on the same page. I will think about the actuator attribute and how it plays into the general signal structure.

As for complex commands I put in "aggregate", which never went anywhere, as a branch attribute to indicate that the entire sub-tree under the branch was to be transmitted atomically. This also translates well to commands that need to set complex values. 

To expand on your mood lighting theme:
```
- MoodLighting.CycleTime
  type: Uint8
  description: Cycle time in seconds. 0 = off

- MoodLighting.StartColor
  type branch:
  description: The color to begin the cycle with
  min: 1
  max: 5

- MoodLighting.StartColor.Red
  type: Uint8
  description: Red intensity
  min: 0
  max: 255

- MoodLighting.StartColor.Green
  type: Uint8
  description: Green intensity
  min: 0
  max: 255

- MoodLighting.StartColor.Blue
  type: Uint8
  description: Blue intensity
  min: 0
  max: 255

- MoodLighting.EndColor
  type: Uint8
  description: The cycle to end the cycle with
  min: 1
  max: 5

- MoodLighting.EndColor.Red
  type: Uint8
  description: Red intensity
  min: 0
  max: 255

- MoodLighting.EndColor.Green
  type: Uint8
  description: Green intensity
  min: 0
  max: 255

- MoodLighting.EndColor.Blue
  type: Uint8
  description: Blue intensity
  min: 0
  max: 255
```

The C-code to set the seat mood-lightning:
```
//
// The typedef MoodLighting below is auto-generated by the VSS tool kit
//
typedef struct _MoodLighting {
    uint8_t CycleTime;

    struct {
        uint8_t Red;
        uint8_t Green;
        uint8_t Blue;
    } StartColor;

    struct {
        uint8_t Red;
        uint8_t Green;
        uint8_t Blue;
    } StopColor;
} MoodLighting;

MoodLighting mood_arg = {
    .CycleTime = 120, // Seconds
    .StartColor = {
        .Red = 255,
        .Green = 0,
        .Blue = 0
    },
    .StopColor = {
        .Red = 0,
        .Green = 0,
        .Blue = 255
    },
};

...
set_signal_by_name("MoodLighting", &mood_arg);
```

Once the command above has been sent, we can then monitor the current color by subscribing to the aggregate branch of `MoodLighting.CurrentColor.[Red,Green,Blue]`.

We are also exploring a design strategy for signals vs. remote procedure calls, but that is somewhat outside the topic of the current issue. Tell me if you know more about that.

danielwilms, 2018-12-10:
> Once the command above has been sent, we can then monitor the current color by subscribing to the aggregate branch of MoodLighting.CurrentColor.[Red,Green,Blue].

This looks pretty interesting and in the coloring example it works well. The only thing I see is, that IMHO it's somewhat orthogonal to the tree structure we have right now. If you want to group existing signals in the tree in different ways you have a lot of duplication of data entries.

> We are also exploring a design strategy for signals vs. remote procedure calls, but that is somewhat outside the topic of the current issue. Tell me if you know more about that.

It's definitely interesting and e.g. in LwM2M it's part of the data model. In general it's linked to the protocol examples on top, like VISS. I'd propose to open another issue for actuators vs. methods and discuss that in the W3C automotive WG.

danielwilms, 2021-06-01:
@magnusfeuer, I guess that's what is now realised in the vsc project. Do you want to keep it open, anyways?

erikbosch, 2024-01-04:
No action in last years, closing this one

closed 2024-01-04 by erikbosch