COVESA/vehicle_signal_specification #91: Uniform way of defining enums

danielwilms, 2018-12-11:
Right now we have different notations for enums. Some examples:

**Underscore** ```Vehicle.Drivetrain.FuelSystem.HybridType```: unknown / not_applicable / stop_start / belt_ISG / CIMG / PHEV 

**Space**  ```Vehicle.Drivetrain.Transmission.DriveType```: unknown / forward wheel drive / rear wheel drive / all wheel drive

**Camel case** ```Vehicle.Cabin.RearShade.Switch```: Inactive / Close / Open / OneShotClose / OneShotOpen

I think it would make sense to agree on a notation, so that it's easier and more intuitive for the developer. I'd go for lower camel case, as we have many lower case enums already. Opinions?

danielwilms, 2021-06-01:
related to #239

erikbosch, 2022-01-25:
Related to #399

erikbosch, 2022-08-10:
Considered fixed by the PRs mentioned above

closed 2022-08-10 by erikbosch