COVESA/vehicle_signal_specification #93: Readme file section Data Unit Type not according to SI

PeterWinzell, 2018-12-14:
For example speed and pressure have multiple units definition in list which means that the list is not according to SI units. So, remove that statement.

klotzbenjamin, 2018-12-14:
Indeed, we had a VSS policy on units that stated that signals and attributes should be defined with automotive standards first, and if there is a doubt, SI units.

Si the README file should list not only SI units but also automotive-specific units (e.g. inches for tire dimension)

What do you think @danielwilms ?

PeterWinzell, 2018-12-14:
Sounds good. I think the statement regarding SI units could be removed from
the README file.

danielwilms, 2018-12-19:
> Indeed, we had a VSS policy on units that stated that signals and attributes should be defined with automotive standards first, and if there is a doubt, SI units.

Do you know, which automotive standard is taken?

>> Sounds good. I think the statement regarding SI units could be removed from the README file.

I think, if the agreement is the way Benjamin describes it, we should put it into the README as guideline. As it's written write now, I agree that it's misleading. Should be fixed.

danielwilms, 2021-06-01:
@PeterWinzell, I saw that even after we changed the documentation, the reference to the SI units is still there. Do you want to make a proposal in a PR?

petervolvowinz, 2023-06-27:
Statement corrected. Close this.

closed 2023-06-27 by PeterWinzell