COVESA/vehicle_signal_specification #110: Array support

UlfBj, 2019-06-06:
VSS does not support that a single node represents an array of data. This could be mitigated by the addition of a new property with e. g. the name “elements”. See the example below where a randomly selected (by me) node is first defined as a single value node, and then in the following definition is an uint8 array with five elements.

RearviewMirror.DimmingLevel:
  datatype: uint8
  type: actuator
  unit: percent
  description: Dimming level of rearview mirror. 0 = undimmed. 100 = fully dimmed

RearviewMirror.DimmingLevel:
  datatype: uint8
  elements: 5
  type: actuator
  unit: percent
  description: Dimming level of rearview mirror. 0 = undimmed. 100 = fully dimmed
I proposed this in the W3C automotive WG, with the following discussion found on link below under “Gen2 ideas”.
https://lists.w3.org/Archives/Public/public-automotive/2019May/

danielwilms, 2020-02-11:
I think it's closely related to instances in a leaf. At the end it's an instance without fixed boundaries, like [1,...]

@adobekan we were discussing about it. I think once the instantiation is merged we can make a proposal.

UlfBj, 2020-02-12:
I agree it is closely related, but different:).
The instance model creates separate nodes for every element in the array, while the idea here is to support the case where the all elements of the array are represented by one VSS node. 
A subtle difference in some aspects, but significant in others. 
If the array is handled as an atomic piece of data on the vehicle bus, then the instance model with separate nodes for each element is not appropriate.

danielwilms, 2020-02-13:
> The instance model creates separate nodes for every element in the array, while the idea here is to support the case where the all elements of the array are represented by one VSS node.

But isn’t that an instantiation of a leaf node?

UlfBj, 2020-02-17:
>> But isn’t that an instantiation of a leaf node?

By one node in the tree generated by the tools, not in the YAML tree. 

Another argument for array support that was mentioned in another discussion about this, was that when using separate nodes for an "array" then there is no guarantee that the values associated with these nodes are captured at the same point in time. If you e. g. have three leaves representing x/y/z dimensions of an accelerometer, then read requests to them do not guarantee they were captured simultaneously, not even if read by one request using a wildcard. Representing the accelerometer output with one node (assuming then that the underlying system handles this data a one "atomic" array of values) would provide that guarantee.

UlfBj, 2020-02-20:
There is also a difference in how requests and responses will look when accessing "array data" represented by multiple instance nodes, or by a single array node, see below where W3C Gen2 syntax is used.

Instance model:
GET  x.y.z?$pathEQ*
->
 .., "value": [{"path":"x.y.z.a", "value": 1}<, ..., {"path":"x.y.z.n", "value": N}]

Array model:
 GET x.y.z
 -> 
..., "value":[1, ..., N]

When the array is represented by multiple nodes a the "value" contains an array of path and element value tuples, while when it is represented by a single node the "value" contains an array of the element values.

danielwilms, 2020-02-21:
> was that when using separate nodes for an "array" 
It's just a matter of implementation. If you extend the instances in a leaf with something like:
```yaml
instances:
    - array[1,n]
```
with this, you could actually make them not directly addressable, but treat them as an array and just mark them as a complex type. I think it is better, because you keep all the other information, especially type and datatype. I think it's crucial not overwriting the type. At the end you have an array of acutators, sensors, etc.

> there is no guarantee that the values associated with these nodes are captured at the same point in time. 

There's no guarantee, with an array either. It completely depends on the implementation. When it comes to that, often I got the request that logically independent measurements, and therefore spread over the tree, should be guaranteed to be taken at the same time. The protocol has to handle that IMHO with an id or timestamp or alternatives. 

> When the array is represented by multiple nodes a the "value" contains an array of path and element value tuples, while when it is represented by a single node the "value" contains an array of the element values.

I'm not against the array concept at all, just how it's expressed in the yaml I care about. I think at the end it's a multiplication of a concept. As instances are intended for. Why not extend the instances then?

UlfBj, 2020-02-21:
>> I think it is better, because you keep all the other information, especially type and datatype. I think it's crucial not overwriting the type. At the end you have an array of acutators, sensors, etc.

In my proposal you also keep all the other information, type, datatype, etc. See the example at the top of this issue thread. 

>> There's no guarantee, with an array either. It completely depends on the implementation.
That is true. But this array syntax could be used to indicate that it is the case. 

>> I'm not against the array concept at all, just how it's expressed in the yaml I care about. I think at the end it's a multiplication of a concept. As instances are intended for. Why not extend the instances then?

Now I think we are getting close to a solution:). 
I do not care for a specific syntax, but for a specific functionality. If we can extend the instance model so that the tool output represents an array with one node, possibly with an accompanying "complex" node, then I have no objections. 
Is your proposal above to be understood that the expression "array[1..N]" is interpreted by the tool to generate a single array node as output? I. e. that the word "array" here is a reserved word in comparison to "xxx[1..N]?
There is still a question how the metadata in this node expresses that it represents array data.

danielwilms, 2020-02-21:
> In my proposal you also keep all the other information, type, datatype, etc. See the example at the top of this issue thread.

Sorry, my bad. I've overseen that.

> Is your proposal above to be understood that the expression "array[1..N]" is interpreted by the tool to generate a single array node as output? I. e. that the word "array" here is a reserved word in comparison to "xxx[1..N]?

yes, that would be a solution, or alternatively:

```yaml
instances:
    - [1,n] 
```

> There is still a question how the metadata in this node expresses that it represents array data.
you mean, like your ```elements``` above?

UlfBj, 2020-02-21:
>> There is still a question how the metadata in this node expresses that it represents array data.
you mean, like your elements above?

I just meant that just as there is "metadata" in YAML signalling this, it is needed also in the tool output node representation. 
But maybe the obvious answer to that is that it is there represented with the same/similar syntax as in YAML. E. g. "instances": "[1,N]". 
Something like that is fine with me. 
With that kind of metadata in the node, I see no need for a "complex" node in association with it.

This kind of arrays should in my mind always start with the index zero (=0), so the YAML representation could then be
instances:
    - [n] 
and similar in the tool output  "instances": "[n]"

gunnarx, 2020-05-19:
I would like to consider arrays of X to be a different data type than X, and describe it in the 
`datatype:` metadata

Example (unknown and known size)
```
datatype: uint32[]
datatype: uint32[100]
```
Q: Is an unknown size worth supporting?

SebastianSchildt, 2020-05-19:
I think "unknown"/variable size is worth supporting, i.e. a list of MAC addresses of neighboring V2V-enabled neighbours

danielwilms, 2020-05-19:
@UlfBj Add example

UlfBj, 2020-05-22:
An example use case for array support may be that a client wants to know the battery voltage of each individual battery cell of the complete EV battery pack. With array support a single request to the single node representing the array of these voltage values will do it. 
Also, as described in the comment https://github.com/GENIVI/vehicle_signal_specification/issues/110#issuecomment-588912666, the response will contain a regular array of the values, instead of a more complex array of objects that would be the case if each battery cell was represented by a separate leaf node.

danielwilms, 2020-06-02:
@danielwilms: add PR with example for the documentation.

danielwilms, 2020-06-02:
@gunnarx: what means unknown size in practice? think about possible misuse

SebastianSchildt, 2020-06-05:
DTCs would be another example where variable size lists/arrays would be very useful

PR #159 (2020-07-01): Introduction of Arrays
* Added definition of arrays
* Fixed minor issues in branches

Fixes: #110
commit 3f1dc3d4 (2020-07-01): Introduction of Arrays (#159)

* Introduction of Arrays

* Added definition of arrays
* Fixed minor issues in branches

Fixes: #110
closed 2020-07-01 by danielwilms