COVESA/vehicle_signal_specification #111: Access restriction tagging

UlfBj, 2019-06-06:
For a server to provide access restriction functionality on the nodes of a VSS tree there must be aq way of tagging the nodes with the appropriate level of access restriction. I propose that this is done by introducing a new property, that could be named “restriction”, or alternatively “access”, that informs whether access to the node require an authorization verification before access is allowed. In order not to have to tag all nodes of the tree, a mechanism of inheritance from ancestor nodes could be applied (“if no explicit tag, a child inherits from parent”). 

I proposed this in the W3C automotive WG, with the following discussion found below under “Gen2 ideas”. 
https://lists.w3.org/Archives/Public/public-automotive/2019May/

klotzbenjamin, 2019-06-07:
> I proposed this in the W3C automotive WG, with the following discussion found below under “Gen2 ideas”.
https://lists.w3.org/Archives/Public/public-automotive/2019May/

I think the correct link is this one: https://lists.w3.org/Archives/Public/public-automotive/2019May/0009.html

I think VSS should stick to the scope of a schema and metadata definition for signals/sensors/attributes but you propose multiple ideas:

> there must be aq way of tagging the nodes with the appropriate level of access restriction

I agree. But this should not be in the YAML spec but rather something you specify in your implementation of VSS, at the same time as when you map VSS to your hardware. 

> that informs whether access to the node require an authorization verification before access is allowed

A good practice would be to "force" the usage of an "access" implementation-specific property. If you do not require an access, this should be epxlicitely mentioned there.

> In order not to have to tag all nodes of the tree, a mechanism of inheritance from ancestor nodes could be applied (“if no explicit tag, a child inherits from parent”).

+1

UlfBj, 2019-06-11:
>>  But this should not be in the YAML spec but rather something you specify in your implementation of VSS, at the same time as when you map VSS to your hardware.

I fully agree.

UlfBj, 2019-10-23:
I think it is now a good timing to define a solution to the access restriction tagging, as the W3C automotive spec referring to VSS is defining a comprehensive access restriction solution. I agree with @klotzbenjamin that this tagging should not be done in the YAML spec, but introduced at a later stage, but I also think the details of this later stage should be defined as part of the VSS specification.

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch