COVESA/vehicle_signal_specification #112: Stream node type

UlfBj, 2019-06-07:
A modern vehicle is typically equipped with multiple sensors that produces streams of data, such as cameras, radars, and lidars. VSS is currently not able to support these sensors. I propose that VSS shall be extended with metadata that enables a client to find the streaming source, and initiate a streaming session. To enable this the “type” property is extended with a new enum value “stream”, and a new property containing the URL to the streaming server is added, “server-url”:<url-to-stream-server>. It needs to be investigated if the URL is sufficient information for a client to initiate a streaming session with at least the most common streaming protocols. If not, possibly one or more properties needs to be added to convey the needed information. 
YAML example:
Camera.Rearview.Right:
  type: stream
  server-url: scheme://path-to-stream-server
  description: Right rear view camera.

klotzbenjamin, 2019-06-07:
We discussed this topic several months ago and I believe we remained in a statu quo.

> VSS is currently not able to support these sensors
I rather think that VSS should only provide the schema and metadata about such sensors/signals, and that the type of interaction should be defined somewhere else. More practically, some signals are meant to be streamed (e.g. camera as you said) and other might or might not (e.g. there are different levels of monitoring, which might require streaming of data).

@UlfBj Is my understanding correct that the "type" property would refer to an exclusive "stream" signal type?

UlfBj, 2019-06-11:
>> I rather think that VSS should only provide the schema and metadata about such sensors/signals, and that the type of interaction should be defined somewhere else.

I  think that the information found in VSS should be what is needed for the client to initiate a streaming session. 

>> Is my understanding correct that the "type" property would refer to an exclusive "stream" signal type?
The value "stream" is one of the "type" values, the others being "sensor", "actuator", etc.

danielwilms, 2020-06-02:
Could be done with a 'normal' node, with datatype string, like

```
camera
camera.status (e.g. active/inactive)
camera.uri -> string
```

maybe not sufficient for connection. @UlfBj: more complex use case

danielwilms, 2020-06-09:
Add example to documentation. @danielwilms

danielwilms, 2020-06-09:
camera state

danielwilms, 2020-07-15:
Documentation available [here](https://genivi.github.io/vehicle_signal_specification/rule_set/data_entry/data_types/#data-streams)

Issue resolved.

closed 2020-07-15 by danielwilms