COVESA/vehicle_signal_specification #138: Support instances in base tooling

danielwilms, 2020-02-16:
As @magnusfeuer in #131 suggests, the instances should be supported in the main tooling. Important fix for v2.

UlfBj, 2020-02-17:
Before starting with this important fix, I would like to bring up two things to consider.

1. Instance syntax in YAML.
In the current update of vspec files I found this
>>  instances:
    - Row[1,4]
    - ["Left","Right

Maybe a list syntax like below is a better alternative?
>>  instances: [Row[1,4], ["Left","Right"]]

2. Adding the new field "complex", or adding a new data type "complex".
Every node in the tree will contain the new boolean "complex" field, with an overwhelming majority set to FALSE. 
In the alternative with a new data type "complex", only the nodes which in the other alternative is set to TRUE will be assigned the "complex" data type. This will be functionally equivalent, and eliminate a new field in every node of the tree. 

Further, "complex" nodes are logically closer to branch type nodes than leaf type nodes (as it has the instantiated nodes as children), so I think it is appropriate that it does have a different node type than its instantiated children. It should then also be stripped from other redundant meta data (e. g. data type, unit, etc) that is anyway found in its children.

danielwilms, 2021-06-01:
done and implemented