COVESA/vehicle_signal_specification #147: New EV signals

UlfBj, 2020-03-26:
Geotab has been asked by the W3C automotive group to propose what signals that could be relevant for the EV use case. This resulted in the document on the link below.
https://mail.google.com/mail/u/0?ui=2&ik=80e1234f77&attid=0.1&permmsgid=msg-a:r-4429506179679065946&th=170f9242229c39bf&view=att&disp=inline&realattid=f_k80hx7nn1

Below you will find a proposal for the VSS integration of the EV "signals" being discussed. 
There are some design questions still to answer, such as the resolution of the energy signals, and how to handle counter wrap-around.

- Delivered energy to the battery during charging (kWh).
- Consumed energy from the battery during "non-charging" (kWh).
-  Delivered energy to the battery from regenerative braking. (kWh).
- Battery state of charge. (%)

These must represent accumulated energy over time as I see it. That is, the values are monotonically increasing (until the value hits max possible, when it wraps around to zero).
What resolution do we believe is needed on these kWh values?
Should the number of "wrap-arounds" be stored by the vehicle?

The existing VSS tree contains the following:
Vehicle.Drivetrain.BatteryManagement.BatteryTemperature
Vehicle.Drivetrain.BatteryManagement.LowBatteryLevel
Vehicle.Drivetrain.BatteryManagement.ChargingInlet

I modified it slightly by adding one more branch -  Battery, which changes the existing entries to:

Vehicle.Drivetrain.BatteryManagement.Battery.Temperature
Vehicle.Drivetrain.BatteryManagement.Battery.LowLevel
Vehicle.Drivetrain.BatteryManagement.ChargingInlet

The proposal for the new EV signals below uses uint32 integers to hold the values, which have the unit Wh. Is the resolution Wh sufficient?
As it says in the presentation that a typical yearly load is about 4000kWh, then a uint32 would take about 1000 years before wrapping around.
For these vehicles there would be no need to keep track of an eventual wrap-around, but maybe for heavy trucks?
It would be sufficient with one wrap-around counter, as  ChargeEnergy and  ConsumedEnergy will not differ significantly. A uint8 should cover the needs.

- Vehicle.Drivetrain.BatteryManagement.Battery.ChargeEnergy:
  datatype: uint32
  type: sensor
  unit: Wh
  description: The accumulated energy delivered to the battery during charging.

- Vehicle.Drivetrain.BatteryManagement.Battery.ConsumedEnergy:
  datatype: uint32
  type: sensor
  unit: Wh
  description: The accumulated energy leaving HV battery for propulsion and auxiliary loads.

- Vehicle.Drivetrain.BatteryManagement.Battery.BrakingEnergy:
  datatype: uint32
  type: sensor
  unit: Wh
  description: The accumulated energy delivered to the battery from regenerative braking.

- Vehicle.Drivetrain.BatteryManagement.Battery.StateOfCharge:
  datatype: uint8
  type: sensor
  unit: percent
  min: 0
  max: 100
  description: The current charging level as a percentage of original battery performance.

klotzbenjamin, 2020-03-26:
I agree with the new branch.

> The proposal for the new EV signals below uses uint32 integers to hold the values, which have the unit Wh. Is the resolution Wh sufficient?
As it says in the presentation that a typical yearly load is about 4000kWh, then a uint32 would take about 1000 years before wrapping around.
For these vehicles there would be no need to keep track of an eventual wrap-around, but maybe for heavy trucks?

Is it relevant to use the SI unit (Wh) while we have a usual de facto energy standard (kWh)? Following the best practices for units in VSS we should use preferably automotive standards and then SI. In this case, kWh would be preferred, and there is no longer a reasonable risk of wrap-around. The only question is the resolution needed (and I do not have the answer), unint32 of maybe float?

Concerning the YAML, I can provide feedback with an inexperienced eye:
```
Vehicle.Drivetrain.BatteryManagement.Battery.ChargeEnergy:
    datatype: uint32
    type: sensor
    unit: Wh
    description: The accumulated energy delivered to the battery during charging.
```
I would rename `ChargeEnergy` to `AccumulatedChargeEnergy` to fit the description. To my understanding, this is the total amount of energy delivered to the battery during the entire lifecycle of the vehicle.
```
Vehicle.Drivetrain.BatteryManagement.Battery.ConsumedEnergy:
    datatype: uint32
    type: sensor
    unit: Wh
    description: The accumulated energy leaving HV battery for propulsion and auxiliary loads.

Vehicle.Drivetrain.BatteryManagement.Battery.BrakingEnergy:
    datatype: uint32
    type: sensor
    unit: Wh
    description: The accumulated energy delivered to the battery from regenerative braking.
```
Same comment as before for the name and description.

UlfBj, 2020-03-26:
>>  In this case, kWh would be preferred, and there is no longer a reasonable risk of wrap-around. The only question is the resolution needed (and I do not have the answer), unint32 of maybe float?

If we go with kWh then using an integer is problematic if an average yearly value is 4000kWh. To get sufficient resolution the data type must be float. With float the need for a wrap-around counter also goes away. 
So I agree we should go with kwh and float.

>> I would rename ChargeEnergy to AccumulatedChargeEnergy to fit the description. To my understanding, this is the total amount of energy delivered to the battery during the entire lifecycle of the vehicle.

I agree. 
Rename to 
AccumulatedChargeEnergy
AccumulatedConsumedEnergy
AccumulatedBrakingEnergy

danielwilms, 2020-03-31:
@danielwilms, @magnusfeuer check with internal signal list. Additional signals later on.

danielwilms, 2020-04-14:
@UlfBj: create PR

danielwilms, 2020-06-10:
PR merged. Issue can be closed.

closed 2020-06-10 by danielwilms