COVESA/vehicle_signal_specification #148: vspec2franca output not complete

kdopen, 2020-03-26:
I tried running the output of `vspec2franca` through `franca_ara_tools` and it wouldn't transform.

There were two initial issues

* No package statement in the generated `.fidl` file
* No definition of `UTF8String` used to declare `VSS_VERSION`

Further, the array declarations seem to initalise fields for signals which are not in the underlying `struct`, for example the struct is defined as

```
struct SignalSpec {
    UInt32 id
    String name
    String type
    String description
}
```
And here's a single signal from the ADAS file

```
{
    name: "CruiseControl.SpeedSet"
    type: "actuator"
    description: "Set cruise control speed in kilometers per hour"
    datatype: int32
    uuid: 35b910b5452f5fa0a5aa6bb630335c49
    unit: km/h
}
```
There are no fields declared in the `struct` for `uuid`, `datatype`, or `unit` and there is no value defined for the `id` field.

I don't know FRANCA that well, so it's possible this is not a problem.

klotzbenjamin, 2020-03-26:
The tools are moving to https://github.com/genivi/vss-tools as part of a current effort (#144). I am not familiar with FRANCE either.

kdopen, 2020-03-26:
So should we move this issue there, or leave it here?

danielwilms, 2020-03-31:
@gunnarx: it really is an issue and has to be looked at. First, mainly PoC usage but useful. Will create new PR.

gunnarx, 2020-03-31:
@kdopen I will get back to you on this.  
You are right and I think there is a completely different translation that should be done in Franca than the one made here.  Something more like this: https://github.com/gunnarx/vehicle_signal_specification/blob/franca_attributes/tools/vspec2franca_attributes.py
(might need a bit of cleanup)

gunnarx, 2020-03-31:
> So should we move this issue there, or leave it here?

No problem to keep it here now.  We will switch over to the other organization relatively soon but the change proposal #144 has not been closed yet.

gunnarx, 2020-06-02:
See above link for the implementation proposal

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch