COVESA/vehicle_signal_specification #149: Making use of YAML specification

danielwilms, 2020-03-31:
Right now, every file is specified as a plain list, which creates some issues:
1. Instead of an object, each spec file is defined as a list of single objects
2. Nesting is impossible, which leads to repetition in the definition
3. Creates `None` types

Here's one example (it comes from the real spec, but is changed a bit to cover more cases):
```yaml
#
# Steering Wheel
#
### ** The leading '-' creates a list of objects **
- SteeringWheel: ### **This line creates a None type**
   instances:
     - Test[1,2]
     - Test[1,2]
  type: branch
  description: Steering wheel signals

### ** let's assume this should be a list **
- SteeringWheel.Angle:
  datatype: Int16
  type: sensor
  unit: degrees
  description: Steering wheel angle. Positive = degrees to the left. Negative = degrees to the right.

### ** Depth by .-separated string **
- SteeringWheel.Tilt:
  datatype: uint8
  type: actuator
  min: 0
  max: 100
  unit: percent
  description: Steering wheel column tilt. 0 = Lowest position. 100 = Highest position.

- SteeringWheel.Extension:
  datatype: uint8
  type: actuator
  min: 0
  max: 100
  unit: percent
  description: Steering wheel column extension from dashboard. 0 = Closest to dashboard. 100 = Furthest from dashboard.
```
Using `pyyaml` this results into the string beneath, and you can see actually all three problems from above.

```
[{'SteeringWheel': None, 'type': 'branch', 'instances': ['Test[1,2]', 'Test[1,2]'], 
'description': 'Steering wheel signals'},
 {'SteeringWheel.Angle': None, 'datatype': 'Int16', 'type': 'sensor', 'unit': 'degrees', 
'description': 'Steering wheel angle. Positive = degrees to the left. Negative = degrees to the right.'}, 
{'SteeringWheel.Tilt': None, 'datatype': 'uint8', 'type': 'actuator', 'min': 0, 'max': 100, 
'unit': 'percent', 'description': 'Steering wheel column tilt. 0 = Lowest position. 100 = Highest position.'}, 
{'SteeringWheel.Extension': None, 'datatype': 'uint8', 'type': 'actuator', 'min': 0, 'max': 100, 
'unit': 'percent', 'description': 'Steering wheel column extension from dashboard. 0 = Closest to dashboard. 100 = Furthest from dashboard.'}]
```

The new proposal could change the original spec like the following:
* Don't use top level lists
* Nest objects, if there is a hierarchy
* Use Yaml lists to actually describe lists

```yaml
#
# Steering Wheel
#
### ** Defines a real object **
SteeringWheel:
  type: branch
  instances:
    - Row[1,2]
    - test
  description: Steering wheel signals
  
  ### ** Depth through indent **
  Angle: 
    - datatype: Int16 ### ** list defined through standard yaml (would be possible in the version above as well **
      type: sensor
      unit: degrees
      description: Steering wheel angle. Positive = degrees to the left. Negative = degrees to the right.

  Tilt:
    datatype: uint8
    type: actuator
    min: 0
    max: 100
    unit: percent
    description: Steering wheel column tilt. 0 = Lowest position. 100 = Highest position.

  Extension:
      datatype: uint8
      type: actuator
      min: 0
      max: 100
      unit: percent
      description: Steering wheel column extension from dashboard. 0 = Closest to dashboard. 100 = Furthest from dashboard.
```

Tis results into:
```
{'SteeringWheel': {'type': 'branch', 'instances': ['Row[1,2]', 'test'], 
'description': 'Steering wheel signals', 
'Angle': [{'datatype': 'Int16', 'type': 'sensor', 'unit': 'degrees', 'description': 'Steering wheel angle. Positive = degrees to the left. Negative = degrees to the right.'}], 
'Tilt': {'datatype': 'uint8', 'type': 'actuator', 'min': 0, 'max': 100, 'unit': 'percent', 'description': 'Steering wheel column tilt. 0 = Lowest position. 100 = Highest position.'}, 
'Extension': {'datatype': 'uint8', 'type': 'actuator', 'min': 0, 'max': 100, 'unit': 'percent', 'description': 'Steering wheel column extension from dashboard. 0 = Closest to dashboard. 100 = Furthest from dashboard.'}}}
```

What's still open how to handle includes, but I would handle that in a separate issue.

gunnarx, 2020-04-06:
In my opinion, yes, each signal must be an independent object in each file.
So therefore the '-' should be removed
```
- SteeringWheel: ### **This line creates a None type**
```
becomes
```
SteeringWheel:
```

Then, for nesting I think it would be acceptable to have both:

1. According to Daniel's example above
```
SteeringWheel:  # Is really <something>.SteeringWheel - the <something> is defined by where this file was inserted into the tree.
  type: branch
  ...  
  ### ** Depth through indent **
  Angle:    #  Is really <something> SteeringWheel.Angle, i.e. a submode to SteeringWheel
```
but also:
2. Fully or partially qualified names:
```
SteeringWheel:
   type: branch
   ...
# No indentation...
SteeringWheel.Angle:
   ...
SteeringWheel.Tilt:
   ...
```
This is more similar to what VSS is today.   Are both of these possible to allow at the same time?

gunnarx, 2020-04-07:
Actually on second thought I'm not sure I accept this:

```
SteeringWheel:  # Is really <something>.SteeringWheel - the <something> is defined by where this file was inserted into the tree.
  type: branch
  description:  This is the steering wheel
  ### ** Depth through indent **
  Angle:
     type: signal
     description: Measured angle of steering wheel from 0 to 540 degrees
     <more metadata for Angle>
```
Because it puts a subnode (Angle) on the same level as metadata.  This does not match, I feel:
  - the metadata topic names are "fixed" and known beforehand, and the same for all (most) nodes, (depending on layers and stuff of course)
  - But the signal name is unknown name until it suddently appears here
  - Signal definitions and metadata do not feel like the same type of information?
  - How would you be 100% sure which is a sub-node definition and which is just metadata?  And avoid clashes - you might have a "metadata" named Angle?

So make it like a consistently traversable model and guarantee no collision you would normally do something like putting the children under a particular metadata topic so you know where to find them:

```
Steering: 
   type: ...
   description: ...
   children :  
        - Angle:
          type: foo
          description: bar
```
but going down this road looks ugly to me.  It makes the children into a list / array which we seemed to want to avoid in the first place, and all of this does not seem to scale very nicely.  Can we find another way?

gunnarx, 2020-04-07:
>    Don't use top level lists

Agree in principle, unless we're forced to, to be able to have more than one signal per file and also be able to parse the file.   

>    Nest objects, if there is a hierarchy

Also desirable, but in my opinion as an optional addition to a flat list.
But as shown above, I'm not sure that ends up being completely clean.

>    Use Yaml lists to actually describe lists

Yes.

Overall make sure we consider the "deployment file" we have talked about before.  It was preferrable there, some argued, to have a single file for *all* signals.    If that single file is to be coded as a VSS layer, then we must allow a reasonable way to allow valid VSS-formatted files (which layers are) to include every single signal.  If the format requires that to be written as a tree with one new indentation level for each branch level, then it becomes unmanageable.

danielwilms, 2020-04-07:
> Because it puts a subnode (Angle) on the same level as metadata. This does not match, I feel:
>
> * the metadata topic names are "fixed" and known beforehand, and the same for all (most) nodes, [...]
Signal definitions and metadata do not feel like the same type of information?
> * How would you be 100% sure which is a sub-node definition and which is just metadata? And avoid clashes - you might have a "metadata" named Angle?

I see, what you're saying, but I don't fully agree, thinking about it. as sugesstion:
* There can be a rule that all meta-data comes first
* a node is defined, by type, name and description. that sets a node apart from the meta-data
* there could be a keyword `meta`, which contains a dictionary of the meta information, for all non-mandatory fields

```
SteeringWheel:
  type: branch
  description: Steering wheel signals
  meta:
      instances:
        - Row[1,2]
        - test
 
  ### ** Depth through indent **
  Angle: 
    - type: sensor
      description: Steering wheel angle. Positive = degrees to the left. Negative = degrees to the right.
      meta: 
        datatype: Int16 ### ** list defined through standard yaml (would be possible in the version above as well **
        type: sensor
        unit: degrees
 ```

> Overall make sure we consider the "deployment file" we have talked about before. It was preferrable there, some argued, to have a single file for all signals. If that single file is to be coded as a [...]

I saw the layers, as an instance between spec and 'compilation'. I just thought, if it doesn't conflict, it's just fine to use the normal spec for the additional meta-data as well. that would be something we have to think about. but first thought is, that the `meta` object would support that.

gunnarx, 2020-04-07:
I don't understand this answer.   You agree, disagree, or did not understand me?

> I saw the layers, as an instance between spec and 'compilation'. 

I don't really understand that sentence.  Anyway, if it helps I see layers as "source code", just like the standard VSS yaml files (see definition below).  Combining **all** the files (using tools that read all the files) will give you the combined source of truth of the database of signals.  Not sure if this helps?

> I just thought, if it doesn't conflict, it's just fine to use the normal spec for the additional meta-data as well.

If you mean that you could take what I would call the "standard or main VSS files", [these files for example](https://github.com/GENIVI/vehicle_signal_specification/tree/master/spec/Cabin),
and add your own metadata in those, then yes, that is possible in theory (and should be supported by tools).   It's a choice each implementer can make, what should be put into a separate layer instead of a main file.  But we have already brought forward various use cases where it would be inconvenient to edit the main VSS file.   I don't quite get your point as a response to what I wrote.  VSS layers are needed, right? wrong?.  Are just confirming that it **is** possible to add your own metadata to the "main VSS files" (in theory)?  Indeed it is.  

> that would be something we have to think about. but first thought is, that the meta object would support that.

I had a hard time understanding. What is "the meta object" and what should it support?  New, custom, metadata?  I'm quite sure we have already agreed on that several times?  Or?

So, I thought I was clear but probably not.   Let me try a different way with separate facts:  
See if you agree with all this:

* All VSS files (including layers or not layers) should be possible to be written in **all** officially supported formats for VSS files.  (By that I mean files that are compliant with the syntax rules we define for VSS-files)

* A layer is a VSS file.  It has the same formatting rules.  (I'm still holding a door a little bit open that a "layer" could be something different also, but at minimum it *is allowed to be* a normal VSS formatted file).

* **You** described before a "deployment file" that was a single file including *all* signals.

* A single file with all signals (i.e. the whole tree) will IMHO be unmanageable if it can ONLY be written using indentation syntax.   

* Therefore, I say, it should be possible to also use a flat syntax which uses the full path (or partially full, depending on file and context):

This is the flat syntax:
```
Cabin.Sunroof.Position:
     new_metadata : foo
Cabin.Sunroof.Switch
     new_metadata: bar
```

Partial pathis also allowed by VSS today.  (For example the file Cabin.vspec defines things that are implicitly underneath Cabin namespace), and the namespace is controlled by the include statements, so that must also be included in the discussion.

gunnarx, 2020-04-07:
> There can be a rule that all meta-data comes first

One problem, only practical but still.  If you run the file through a processor, like pretty-printer or any program that just parses the file and then prints it out again, then the order is lost.  You probably get the printed in alphabetical order.  We are basically defining an associative array (dictionary, hash-map, different names) and those are most often not keeping track of the order of keys - the order is undefined.

gunnarx, 2020-04-07:
> a node is defined, by type, name and description. that sets a node apart from the meta-data

I have always thought of all the fields as metadata.  datatype and type and description are metadata. Do you have a different definition?  This ought to be written down formally.

> there could be a keyword meta, which contains a dictionary of the meta information, for all non-mandatory fields

Perhaps you rather meant  "the _non-mandatory_ meta-data" then?

> there could be a keyword meta, which contains a dictionary of the meta information, for all non-mandatory fields

This seems analogous to my example of children: but the opposite.  Yes, this is more reasonable than children: containing the nodes.  It still adds a complication however.

But since you are not putting **all** the metadata under meta: there is still some possible name clash.  I think we need some additional rule.  For example say that all node names have Capitalized names and all meta data have small letters only (not unreasonable I think).  Otherwise you could never have a sub-node named Type which clashes with type, as an example.

magnusfeuer, 2020-04-08:
I made an implementation of `vspec.py` that parses both dictionary-style specs and the existing list-based format. The dictionary specifications are non-indented, retaining the current dot-notation naming format.

https://github.com/magnusfeuer/vehicle_signal_specification/tree/issue-149-flat

`tools/delist.sh` Converts the existing vspec file tree from list-based to dictionary based. 
It also converts `#include` statements to regular YAML elements, because `#include` needs to die.

To test:

1. Run the CSV generator on the current list-based spec:
```
./tools/vspec2csv.py -i:spec/VehicleSignalSpecification.id -I ./spec ./spec/VehicleSignalSpecification.vspec old.csv
```

2. Convert the spec files from list to dictionary (and fix `#include`)
```./tools/delist.sh spec```

3. Run the CSV generator against the dictionary-based spec.
```
./tools/vspec2csv.py -i:spec/VehicleSignalSpecification.id -I ./spec ./spec/VehicleSignalSpecification.vspec new.csv
```

4. Verify that the output from both runs are the same:
```
diff old.csv new.csv
```

You can also check the CSV files against one generated by the original code that I forked from GENIVIs repo:

```
git checkout master
git checkout spec  # Undo the list-to-dictionary conversion done by delist.sh
./tools/vspec2csv.py -i:spec/VehicleSignalSpecification.id -I ./spec ./spec/VehicleSignalSpecification.vspec original.csv
diff original.csv new.csv
```

magnusfeuer, 2020-04-23:
After talking to @danielwilms on Tuesday, and re-reading this issue thread, IMHO we should have custom meta data stored under a dedicated element  (`metadata:`)

If we stay with the flat, non-nested structure of VSS files (as implemented in my proposal above) we will not run into name clashing between children and meta data. The internal structure built up by vspec.py stores all children in a dedicated 'children' element, as can be seen in the generated JSON signal spec.

Given the assumption that we will not do nested signal structures for now, are you (@gunnarx, @danielwilms, etc) ok with me putting in a pull request into the VSS repo for the code above?

danielwilms, 2020-04-24:
@magnusfeuer, fine for me. Go ahead. We can take the discussion from there. We can open another one for the nesting issue.

magnusfeuer, 2020-04-27:
Pull request submitted: https://github.com/GENIVI/vss-tools/pull/4

danielwilms, 2020-05-19:
@danielwilms: check PR, @UlfBj ?

gunnarx, 2021-05-25:
The PR GENIVI/vss-tools#4 was merged.  The rest of the discussion seems to be on how to instantiate which I think is also completed. #166 has related info.

PR #296 (2021-06-22): Changed VSS elements from YAML list to dictionary
- Changed list entries in spec file with kudos to @magnusfeuer for providing the script
- Fixed documentation

Fixes: #149

Depends on: GENIVI/vss-tools#68
closed 2021-06-22 by gunnarx