COVESA/vehicle_signal_specification #150: vss-tools: Write a validator

gunnarx, 2020-04-07:
There are some conventions and rules that ought to be checked, such as capitalization etc.

klotzbenjamin, 2020-04-09:
Would this be to check the validity of a vspec file? I agree that it makes sense.

From the [doc](https://genivi.github.io/vehicle_signal_specification/), we have a set of rules and practices, I would list:
* UpperCamelCase for node names
* Node relationships: except for the "root" Vehicle node, every node has a parent. A typo in a YAML can remove this link
* Node types: one of branch, attribute, sensor, actuator, stream
* Data types, we have a similar list
* Units: we have a similar list, but this one might change with new signals.

I would sum up by saying that we have to check the tree structure, the nodes consistency and the entries of each node.

gunnarx, 2020-04-14:
I agree with all the checks above.

> Units: we have a similar list, but this one might change with new signals.

Despite this I think it is worth it to check units and warn about unknown units.  Also, common "mistakes" might be highlighted, assuming we can agree on what the standard is.   E.g. kilometer -> km (assuming this is preferred)

danielwilms, 2021-06-01:
related to #239, needs best practice documentation as well

erikbosch, 2021-12-21:
After merging https://github.com/COVESA/vss-tools/pull/128, what is left to validate?

erikbosch, 2024-02-12:
Quite much implemented now, closing this one

closed 2024-02-12 by erikbosch