COVESA/vehicle_signal_specification #156: Enums Part 1: Rename enum

danielwilms, 2020-05-05:
Right now, `enums` are used as `input restrictions` with the datatype, which is defined in the leaf. The topic of enums caused quite a bit of confusion. So I'd like to break the discussion into two:

Can we agree on renaming of the currently supported concept from `enum` to `acceptedValues` or are there any objections?

danielwilms, 2020-05-12:
- mapping of integer might be a missing feature
- acceptedValues might be as confusing (e.g. regEx)
- check use-case as an example
- backward compatibility important

gunnarx, 2020-06-23:
I am opposed to this changed because I think the VSS model needs a proper enum type and I don't really like either the purpose or name of "acceptedValues".
 
Actually, I'm not sure if what you are saying _used as "input restrictions"_ is more or less what an enum actually is?

I'm also likely to be OK with breaking backward compatibility if the resulting VSS model is better. Specifically I would accept redefining some signals that already exist in VSS standard catalog - for example change those who use the old enum concept are defined to use a new better enum concept.

A more explicit question, IMHO, is if particular **enum types** should be possible to define once, and reused in several places.  Instead of being defined within the node that uses them (potentially with repetition if another node uses the same).   This, however, is likely to bring us to a more general complex-data-type definition discussion.

danielwilms, 2020-08-04:
* Generalized concept is preferred
* Enum better than alternatives
* Maybe integer-mapping could be a good extension

@gunnarx creates example

erikbosch, 2022-01-25:
Can this issue be closed now @danielwilms , as we have merged https://github.com/COVESA/vehicle_signal_specification/pull/381

closed 2022-08-10 by erikbosch