COVESA/vehicle_signal_specification #157: Enums Part 2: Enum concept needed besides input restrictions?

danielwilms, 2020-05-05:
In #156 the renaming of enums is discussed. I'd open this issue here to discuss whether an enum concept beyond that is needed. As it's handled differently in different programming languages, here what wikipedia says:

> In computer programming, an enumerated type (also called enumeration, enum, or factor in the R programming language, and a categorical variable in statistics) is a data type consisting of a set of named values called elements, members, enumeral, or enumerators of the type. 

I would argue, that's what VSS with branches and leaves is all about. I don't see a big value adding it as an additional concept. Opinions?

gunnarx, 2020-06-23:
>  would argue, that's what VSS with branches and leaves is all about

It is the ability to use enum as a type, and the enumerations as values.  I don't see how this can be mimicked conveniently by VSS branches and leaves.

Example:

NOTE I'm not proposing this exact syntax, just proving the model here:

```
FavoriteFruit:
   - type :  enum { "banana", "apple", "strawberry" }
```

If I fetch the value of FavoriteFruit I get its current value, which is one of the three predefined values.  

It seems not convenient to model this in a VSS tree, or how would you so it?

Let's try with separate nodes as the enumerations:
```
FavoriteFruit.Banana
   - type: boolean
   - description: Returns true if favorite is banana, otherwise false
FavoriteFruit.Apple
   - type: boolean
FavoriteFruit.Strawberry
   - type: boolean
```

You have to fetch all three, or fetch them as a group and see which one is "true".  It's very clumsy, and of course increasing the more possible values there are.

or you use a string?
```
FavoriteFruit
   - type: string
   - description: Returns a string "banana", "apple" or "strawberry" to indicate the fruit.
```

That option throws away the strict typing.

Maybe there is some other way I don't see.

I think there have been a few more complex examples brought up before, such as the desire to have a signal that is an array of enum values.

gunnarx, 2020-08-04:
Some observations.

1. If generalized, then enums are (in my understanding) limiting the possible assigned values to the given list.  As such, I think enums (in general sense) are actually "input restrictions".  I don't think the current definition is all that far from reality.

2. Enums have an underlying type in which the limited list of values could be stored.  Traditional C-programming enums have (only) an integer type.  In C there are various rules and also some implementation-defined behavior for which _type_ of integer type is used in C and C++.  In modern C++, it is possible to define which type of integer to use.

I think other languages might have non-integer enum types, so maybe it should not be assumed to be integer.

3. Currently it seems that .vspec files have only enum items using **string** type, with a list of allowed strings. With the general definition of enum, I think it would be fine to allow type string and a limitation of strings that can be used.

4.  Maybe some of the .vspec enums using string should really be defined as integer. It is hard to know 

5. There is no way today to assign specific integer values to each symbolic named value.

6. I am not too happy with "input restriction".  I think continue using enum or enum_values or similar name would be quite OK.

gunnarx, 2020-08-04:
An example from current vspec:

```
enum: [ "Inactive", "Close", "Open", "OneShotClose", "OneShotOpen", "TiltUp", "TiltDown" ]
```
This one uses string (see above) and would act rather as an input restriction - only those string values are expected.

Proposal for integer variant:
```
type:  int16 
enum: [ Inactive, Close, Open ]
```

For integer-based underlying types (only) it should be allowed to define the explicit value to use:
```
type:  int16 
enum: [ Inactive = 1, Close = 10, Open = 20 ]
```
The above is valid YAML according to one syntax checker I found, but it should be noted that the semantic meaning of it is that each item in the array is one YAML item.  In other words you can view it as:
    "Inactive = 1", "Close = 10", "Open = 20"

The assignment is not part of the YAML semantic meaning, and it seems it will be required for generators to look into each string, parse it to find the equal sign and interpret the integer value.

YAML cleanliness would require:
```
type:  int16 
enum: [ Inactive : 1, Close : 10, Open : 20 ]
```
whereas the former seems more familiar and convenient.  
Which one is your preference?

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch