COVESA/vehicle_signal_specification #163: Avoiding unsigned-signed problem with integers...

gunnarx, 2020-06-16:
It is well known that the mapping of signed and unsigned integer types can cause problems in certain programming environments.  Some environments, prominently Java, does not have unsigned types.

This makes mapping difficult since the maximum value of an unsigned integer does not fit into its corresponding signed integer sibling.  It might be good to proactively address this issue using  documented requirements for the VSS.

Since the collaborative project that  defines the standard+VSS database is in control of its contents, and equally, whoever may be defining a proprietary extension to VSS is also in control of that definition, we could document a set of agreements on how to handle this problem.

Might the following be a useful proposal?

1. **Recommend** how to map unsigned to signed-only environments as follows:
   - For any UINT (except UINT64) that appears as datatype in VSS, choose the next _larger_ signed integer when representing this in the target programming environment:

     e.g.  UINT8 -> signed int 16,  UINT -> signed int 32, and so on.

This is a common and well-known solution but it then breaks down when you get to UINT64 -> ?

2. (THIS IS THE BIG CHANGE, but please read to the end\*\*):   **Require** that values transferred by a VSS signal that uses type UINT64 must have a **maximum value** of 2^63-1, (not the typical 2^64-1), by definition.  If this is required by the VSS specification then it would allow implementations to safely implement a mapping from unsigned-64 to signed-64 without risking that it is out-of-specification.  Values that are beyond scope might be handled by logging errors, or by some undefined (but reasonable) behavior.

3. For the entity that controls the definition of the VSS (standard or proprietary extension), recommend the following when defining new VSS signals:

A) For any in-car data that arrives to the VSS translation point represented by a uint64, of course this might not be open for redefinition easily.  The key is to figure out whether the full range of the 64-bit value is ever used. (In essence the agreed specification/contract of this data item must be confirmed!)   If it can be confirmed to be within the 2^63 limit then signal definition can proceed without issue because the principle in 1) can be used on the receiving side.   In other words, the internal system design may choose to forward the requirement that VSS has, to say that the sent value must be maximum 2^63, else the behavior is undefined.   Or it may simply enforce this limit at the translation point where it becomes a VSS signal, if that is appropriate.

B) Furthermore, the next alternative is to, knowingly, define the signal to be signed-int-64 in VSS and transfer the value with the explicit knowledge that it must be bit-converted on the receiving side according to standard 2-complement rules of signed integers.  In other words, it would be sending a unsigned 64 bit value that is deliberately coded as the equivalent "signed value" that would have the same bit encoding.   The translation would be akin to an **unchecked** C-programming cast like:
*(int_64)  &my_unsigned_64_bit_value, and the opposite on the receiving side.
The receiving side can extract the value from the signed-integer bit representation by 2-s complement conversion.

If this trick done with explicit documentation for the signal in question, then it should be unambiguous and it may be an acceptable solution.  But it must be explicitly documented!

C) A less _hackish_ solution for the VSS signal designer is that for values known to potentially be larger than 2^63, to represent these using a decimal string in the VSS database.  In the end, those that are known to potentially be that large, might be relatively few.

D) Yet another solution is to represent the value using a constructed 128-bit (signed) value by encoding it as a 2-value array type in VSS.  The array would include two 64 bit integers (upper and lower part).  Again, documentation handles the explanation of what is actually being transferred.

Note: On the receiving programming environment  (Java or something else) the received value from any of the above could be put into a signed 128 bit integer if that was supporte (which is rare).   But any environment can use a "BigInt" type of library  that allows for any sized values, or just internally the use two integer approach, upper and lower, to store and represent values of any size.

-----

\*\* In the end, when taking a step back and looking at the above, the limitation to 2^63-1 maximum value for UINT64 boils down to that any value above that would be defined as illegal value.  The exact same thing would be achieved by instead using a **signed** INT 64, and (in the 
documentation) stating that _negative_ values are illegal.
Therefore, perhaps the right conclusion ought to be instead, that UINT64 shall not be allowed as a data type in VSS and to recommend using the signed 64 instead, where appropriate.  For any data that does not fit into this limitation -- all of the above described tricks to handle the issue are equally valid in that case.

SebastianSchildt, 2020-06-16:
I won’t disagree with these observations, but I would argue  it is not a “VSS problem”. There are millions of languages and technologies, and we should in our design not try to accomodate a few chosen ones. In the data model we should clearly define uses signed/unsigned 8/16/32/64 OR better yet, allow arbitrary length integers, setting arbitrary min/max and let the implementation figure out the rest

I mean for total freedom you can encode numbers as SDNV or or some TLV format, but again, I think not a data model issue.

danielwilms, 2020-06-24:
> allow arbitrary length integers, setting arbitrary min/max and let the implementation figure out the rest

From my view that would be the cleanest solution. That would mean at the end a heavier usage of min/max than we do today.

danielwilms, 2020-07-21:
Create PR @danielwilms min/max + default value for integers.

gunnarx, 2020-08-04:
Last week we sort of "forced" through the merge of https://github.com/vss-tools/pull/7
because the fundamental part of it is something everyone  wanted but it actually had an unanswered open question:

It does the following mapping without any explanation!
```
    UINT64 = "Int64"
```
So (as I pointed out in the reference) it is related to this.
Shall we file that as a separate "bug" now or what?

>  better yet, allow arbitrary length integers.   I mean for total freedom you can encode numbers as SDNV or or some TLV format, but again, I think not a data model issue.

I see where you are coming from but in my experience, when models become too "pure" or abstract, they can be less useful in practice due to the complexity or extra work involved.  Very frequently programmers will select something else that is more directly suited.   There is also no doubt that VSS was first developed from a more "low-level" heritage (thinking about CAN networks and such), so for that reason I think the named integer sizes should remain in VSS even if it now used as a much more generic data model.   Interestingly, Franca IDL, which is designed to be a _very generic_  IDL, even includes sized integers because they are so prevalent.   This is despite the fact that Franca in theory could have pushed all integer-to-bits mapping into the deployment model, which is heavily used in other areas, but it would just be too much inconvenience to require every integer a definition in both the interface, and the deployment.

You should not be forced to separately define min and max for every integer.  The modeling possibility should be there, but not required, IMHO.  Again, it's a balancing act and I propose practicality over cleanliness here.

> There are millions of languages and technologies, and we should in our design not try to accomodate a few chosen ones. 

Rather I think I am trying to accomodate _all_ of them, by handling the difficulties of some (all others are no problem), but it might be a matter of definition.   A key motivator here is to ensure beforehand that "let the implementations sort it out" does not lead to incompatibilities over a long chain of connected VSS-related technologies, if we can avoid that already at an earlier stage.  No single implementor has control over the whole distance that VSS data may travel from source to sink, as I see it.  What you get out in the other end should be understandable and not corrupted.  I think this is the key value of VSS and we should try to anticipate issues that occur in actual implementation.

> In the data model we should clearly define uses signed/unsigned 8/16/32/64

I prefer this over only un-sized "Integer".  After a lot of back and forth in the introduction my explicit proposal is now only:  Include all the choices _except_ an unsigned 64-bit integer (only signed integer is available for 64-bit).  Also for this case you could use the argument "let the implementation sort it out" (such as if someone, for whatever reason, wants to define data that is really an unsigned, but is forced to use the signed option).

gunnarx, 2020-08-04:
> Create PR @danielwilms min/max + default value for integers.

Not sure I understood what this conclusion was.  Is it to require definition of min/max on every integer? (I'm not a fan of requiring that).  Is it done by adding additional (mandator) metadata fields to the core VSS?

Defining a "default value" - no I'm pretty sure I don't support that, bad experiences from this.  Please let's discuss the motivation once again.

UlfBj, 2020-08-28:
>> Therefore, perhaps the right conclusion ought to be instead, that UINT64 shall not be allowed as a data type in VSS and to recommend using the signed 64 instead, where appropriate. 

I think this should be the solution. Maybe together with a recommendation to use FLOAT if the INT64 limitation is not acceptable.

danielwilms, 2021-06-01:
Should be resolved with https://github.com/GENIVI/vss-tools/issues/8. Will close it for now. If further discussions on `number` with `min`,`max` is needed, let's create that in a separate issue.

closed 2021-06-01 by danielwilms