COVESA/vehicle_signal_specification #165: DTC node missing

UlfBj, 2020-06-19:
@UlfBj : After reading this article
https://otonomo.io/blog/extended-vehicles-a-new-eu-regulation-based-framework-for-car-data-access/ 
where it says that DTCs are some of the data that must be accessible from Sept 2020.
I searched on DTC in the VehicleSignalSpecification.id file, and found the signals below. 
To my understanding none of them provide the DTCs (the freeze DTC being an exception). 
I think we should add DTC to the tree, with the new array datatype (unlimited).
Vehicle.OBD.DistanceSinceDTCClear: 0da628e2c69d561eb86216ddcb6e7b2a
Vehicle.OBD.DriveCycleStatus.DTCCount: 312856f746ff560e8098c19196964d3b
Vehicle.OBD.FreezeDTC: 5b87fae8dda4522aae209ae528960782
Vehicle.OBD.Status.DTCCount: 4afdf65e788c5f69baf682597e69fb67
Vehicle.OBD.TimeSinceDTCCleared: 66ea3984a2585dcdaaf6452eef835c0d
Vehicle.OBD.WarmupsSinceDTCClear: a63ba60721785fc591e3dd067c4dc2ae
The article also says "It should be in standardized formats (while acknowledging that standards are not yet available)". If we want VSS to become the standard for this, then we should at least ensure it meets what is required.

Schildt: Yes, we need DTC, that is always the „Hello world”, when using OBD… I think I already mentioned it as an example in the array discussion, so I agree: This is exactly a usecase for arbitrary length arrays.
So at least we need a list of DTCs as array
(Maybe we want to separate P/U/C.. codes or maybe generic from manufacturer specific, but I suppose a flat list is enough, that is also what OBD gives you)

UlfBj, 2020-06-19:
@gunnarx 
> After reading this article
> https://otonomo.io/blog/extended-vehicles-a-new-eu-regulation-based-framework-for-car-data-access/
> where it says that DTCs are some of the data that must be accessible from
> Sept 2020.

It is a reasonable extension of the requirement already today to provide
standard diagnostics on the physical OBD-2 port.  In the same way, as the
OEMs are implement diagnostics-over-the-air, the wider industry is asking
that the same capability is made available to other third-party companies
involved in the business of car service station, or other.

> I searched on DTC in the VehicleSignalSpecification.id file, and found
> the signals below.
> To my understanding none of them provide the DTCs

I understand your proposal (I think) and comment on it below.  But before
that I am thinking of the DTCs themselves:

I believe many DTCs are  OEM-specific and unlikely to be in the VSS-
standard catalog, but some are also standard.   In any case a tree branch
could be defined and the VSS documentation could explain the ability to
extend it to define actual DTCs, and PIDs/DIDs (data items).

You probably all know this, but just to summarize what we know:
The specification ISO-14229, and related ones, define standard automotive
diagnostics:

14229:1 is the general definition of protocol
14229:2 describes the so called session layer, i.e. general services.
These service definitions appear on each of the buses.
:3 to :7 describe how to apply it on different automotive buses.  Most
interesting perhaps :5 which is for diagnostics over IP.

Here is a simple overview page and in the references the link to the
official spec [1]
https://www.automotive.wiki/index.php/ISO_14229

As we can see there it basically only specifies the protocol required to
read DTCs, and other functions.  I don't remember the nature of 0x19 - Read
DTC, if you get all the active ones, or if you input the DTC number you
want to check.  For 0x22 - Read Data by Identifier, I'm quite sure you
input a number and this number/meaning is definitely often proprietary.
(DTCs and DIDs have often been reverse-engineered however so the knowledge
tends to spread but this is not really useful for a stable standard, and
they are still different on car brands)

But there is clearly a set of standard numbers as part of what is
required/standard diagnostics.  Otherwise standardized diagnostics on OBD-2
port would have little value.  Only these, potentially, would be reasonable
to list in a standard VSS catalog, from what I can tell.

Using VSS (+protocols) for this would act only as an alternative to the
Diagnostics-over-IP standard (14229:5).  Even though that protocol is
usable, and used, also in-vehicle, I would imagine this may already used by
some companies "directly".  Meaning that once you have established a
(secure) connection to the car, it might tunnel the DoIP protocol through
it as specified in the ISO specification.

> (the freeze DTC being an exception).

I haven't looked too much at the definition in the current VSS but
Freeze frames are to my knowledge even more proprietary in nature?
And there isn't only one but rather they are associated with certain
faults and read out in the case of that fault happening.  I'm not sure how
to define that reasonably in the VSS.

Again, I'd recommend we create some documentation to show how to define
them in a VSS tree but it would be primarily an instruction for how to do
it, and OEMs and suppliers may take over the task of writing those files...

[1] ISO-14229:1 https://www.iso.org/standard/72439.html

>
> I think we should add DTC to the tree, with the new array datatype
> (unlimited).

I'm concerned with the standard vs proprietary nature here.  What I
remember from your call, your proposal (Ulf & Sebastian?) was that if you
request this node then you get back a list of all DTCs that are currently
"set" or active?  (I'm just guessing "active" is the right word - we must
use real terminology according to 14229 for this).  So on a fault-free car
you would get an empty list.  This is the intention?

This might be valuable but does not fulfil the whole scope of diagnostics-
over-the-air so some might find this quite limited in value, because they
need full DoIP anyway.  If we do add it, it should not be named "DTC" but
something like "ActiveDTCs" or similar, for clarity.

UlfBj, 2020-06-19:
@gunnarx 
> After reading this article
> https://otonomo.io/blog/extended-vehicles-a-new-eu-regulation-based-framework-for-car-data-access/
> where it says that DTCs are some of the data that must be accessible from
> Sept 2020.

It is a reasonable extension of the requirement already today to provide
standard diagnostics on the physical OBD-2 port.  In the same way, as the
OEMs are implement diagnostics-over-the-air, the wider industry is asking
that the same capability is made available to other third-party companies
involved in the business of car service station, or other.

> I searched on DTC in the VehicleSignalSpecification.id file, and found
> the signals below.
> To my understanding none of them provide the DTCs

I understand your proposal (I think) and comment on it below.  But before
that I am thinking of the DTCs themselves:

I believe many DTCs are  OEM-specific and unlikely to be in the VSS-
standard catalog, but some are also standard.   In any case a tree branch
could be defined and the VSS documentation could explain the ability to
extend it to define actual DTCs, and PIDs/DIDs (data items).

You probably all know this, but just to summarize what we know:
The specification ISO-14229, and related ones, define standard automotive
diagnostics:

14229:1 is the general definition of protocol
14229:2 describes the so called session layer, i.e. general services.
These service definitions appear on each of the buses.
:3 to :7 describe how to apply it on different automotive buses.  Most
interesting perhaps :5 which is for diagnostics over IP.

Here is a simple overview page and in the references the link to the
official spec [1]
https://www.automotive.wiki/index.php/ISO_14229

As we can see there it basically only specifies the protocol required to
read DTCs, and other functions.  I don't remember the nature of 0x19 - Read
DTC, if you get all the active ones, or if you input the DTC number you
want to check.  For 0x22 - Read Data by Identifier, I'm quite sure you
input a number and this number/meaning is definitely often proprietary.
(DTCs and DIDs have often been reverse-engineered however so the knowledge
tends to spread but this is not really useful for a stable standard, and
they are still different on car brands)

But there is clearly a set of standard numbers as part of what is
required/standard diagnostics.  Otherwise standardized diagnostics on OBD-2
port would have little value.  Only these, potentially, would be reasonable
to list in a standard VSS catalog, from what I can tell.

Using VSS (+protocols) for this would act only as an alternative to the
Diagnostics-over-IP standard (14229:5).  Even though that protocol is
usable, and used, also in-vehicle, I would imagine this may already used by
some companies "directly".  Meaning that once you have established a
(secure) connection to the car, it might tunnel the DoIP protocol through
it as specified in the ISO specification.

> (the freeze DTC being an exception).

I haven't looked too much at the definition in the current VSS but
Freeze frames are to my knowledge even more proprietary in nature?
And there isn't only one but rather they are associated with certain
faults and read out in the case of that fault happening.  I'm not sure how
to define that reasonably in the VSS.

Again, I'd recommend we create some documentation to show how to define
them in a VSS tree but it would be primarily an instruction for how to do
it, and OEMs and suppliers may take over the task of writing those files...
[... 51 lines cut ...]
possible to avoid it completely, though.   The tree may have (several)
named freeze frame nodes, which deliver data in an anonymous array.  This
is clearly for a proprietary part however, and unlikely to be anything
defined in VSS standard catalog.

In my analysis of "Value exchange formats" I named one concept "Snapshots"
inspired by freeze frames.  The Snapshot is defined as a collection of
values corresponding to different signals that were measured at the same
time.  The intention is basically a "freeze frame" but each part of it is a
signal that is also explicitly defined in VSS terms, instead of the whole
item being an anonymous blob of data.

> So at least we need a list of DTCs as array

Again to be sure, "List of DTCs" is not the actual full list of available
DTCs right?  I mean it is an array of the current active/set DTCs.  We
should clarify this or it becomes misleading.  I had misunderstood myself
for a while...

> (Maybe we want to separate P/U/C.. codes or maybe generic from
> manufacturer specific, but I suppose a flat list is enough, that is also
> what OBD gives you)

If it's just a list of currently active faults then I guess the receiver
can sort out which ones are generic and proprietary.

UlfBj, 2020-06-19:
Schildt
> Again to be sure, "List of DTCs" is not the actual full list of available DTCs right?  I mean it is an array of the current active/set DTCs.  We should clarify this or it becomes
> misleading.  I had misunderstood myself for a while...
Yes, I meant a list of active DTCs. And of course DTC doesn't need to be the only place for some information. An entry in P1999 in the active DTC list can of course happily coexist with Vehicle.Powertrain.Engine->isOnFire=true , which is more "VSS-like" and accessible

>> (Maybe we want to separate P/U/C.. codes or maybe generic from
>> manufacturer specific, but I suppose a flat list is enough, that is
>> also what OBD gives you)

> If it's just a list of currently active faults then I guess the receiver can sort out which ones are generic and proprietary.

Agreed

UlfBj, 2020-06-19:
@gunnarx 
> Yes, I meant a list of active DTCs. And of course DTC doesn't need to be
> the only place for some information. An entry in P1999 in the active DTC
> list can of course happily coexist

I agree in principle.

> with Vehicle.Powertrain.Engine->isOnFire=true , which is more "VSS-like"
> and accessible

What I was after was more that if such a DTC for  a "thermal event" in the
engine :) is actually a standard DTC then it could be explicitly named in
the tree under something like  DTCs.engineThermalEvent

Whatever we choose, I would still strive to avoid duplication of the exact
same information in several places if possible.  Assuming it is the *exact*
same information.  *Similar* could still be useful in some cases.

UlfBj, 2020-06-19:
Schildt
I would argue that " DTCs.engineThermalEvent" is not a good idea. I can explain :)

- Option 1: Have an array of Numeric DTCs as array: the advantage is, the DTC concept is well known and everyone one from Consumer  Android ELM327 Apps to commercial diagnostics testers use it. We can see the "OBD" branch in VSS as an optional "legacy" branch that very closely mirrors what OBD can do, so all apps can be ported easily, and find everything they expect

- Option 2: Put errors/status/condition codes where they belong according to the VSS taxonomy: Vehicle.Powertrain.Something -> That is VSS like

 - Option 3: DTCs.engineThermalEvent ,...   is sort of breaking the VSS taxonomy (it would maybe almost be alike having a "sensors" or "actors" root level node with stuff under it), and DOES NOT help current users of DTCs. If they need to change from parsing "P1999,U0656" to DTC.engineThermalEvent, they could as well use a "real VSS path"

Therefore I would argue to just have a "classic" list of DTC codes in the OBD branch (or someplace else, if we start a "Diagnostics/Self-check branch or similar) and at the same time - if it makes sense - add status information in the "right places" in the tree, even if it means duplication (we could even have a convention like a generic "state" field, where you can walk the tree and find which subtries are "ok", and what is broken).But for DTCs-> Keep it simple and do not surprise users,

UlfBj, 2020-06-19:
@UlfBj 
I support the idea of a "classic" list of DTC codes. 
The article mentions not only "standard" DTCs, but also OEM specific DTCs. Maybe we should have one node for the standard set and one for the OEM specific set?

I agree that this discussion should be moved to a Github Issue.

danielwilms, 2021-05-25:
Node Added, can be closed

closed 2021-05-25 by danielwilms