COVESA/vehicle_signal_specification #166: Instances: Deployment vs. Definition

danielwilms, 2020-07-09:
With the ongoing instance discussion, it seems that the main concern is
how to flatten the tree regarding instances. Putting them into the path,
or at the end. And I see that there's a strong desire to define a unique
identifier for each leaf, including the instances. Within quite some discussions
yesterday, we came at some point to the following conclusions:  
* the flattening is one issue, but not the decisive one.
* we have to answer the question, if the exact number of instances is really 
a question the data-model should answer, or if it's a question of the deployment.

I know, that it doesn't give an answer how to flatten the particular tree,
but I think it has implications on the discussion.

[image]

I tried to distinguish the following:  
* **VSS CORE**: Concerns all vehicles, data entries have fixed identifiers with dot notation
* **VSS DEPLOYMENT**: Through a specific protocol accessible data tree for a
  specific vehicle (type).

From my perspective only the `VSS CORE` should be part of the specification.
This means, that the  number of `instances` for a node is not a concern of the
core specification, as it is dynamic information to a vehicle type. But the fact,
that this specific node is an instance, does.

As I see the core and the deployment as different concerns,
made me argue strongly to flatten it towards the end, as the core tree is stable then.
I think `instance` could then be promoted as a node type should survive, 
as implementations can decide,
how they serve them according to the need of the protocol. Further, the naming
conventions (e.g. `Row`) have to be clear and part of the definition as well.

**But in a nutshell:** The thinking of flattening the instances towards the end is
mainly driven by the wish to keep the core tree stable and readable. So, the
main question is, if the instantiation is a question the core data model should
answer in the first place. That being said, I can follow @gunnarx point, that this
could be the main focus *after* finishing up version 2, including 
deployment/layering. As there are a lot of changes already, which wait to get out. 
For version 2 we could leave this one untouched and flatten
the tree, as it's been done up to now.

What's your opinion: @UlfBj, @SebastianSchildt, @magnusfeuer, @tguild ?

SebastianSchildt, 2020-07-09:
Will write a more detailed answer/suggestion later. I strongly support the notion of separating core taxonomy/tree from deployment information. VSS should tell me how a wheel is modelled, but how many wheels (2-wheeler, car, truck) is deployment issue. That way we can really say vehicles use the "same" VSS without the need to model a - cumbersome- superset.  This implies to have "instance" metadate in a separate file that is processed by our tooling together with the core model. I do believe this would imply "instances" need to become a core vss concept, i.e. the core tree need to be marked, which elements will be instantiated by a deployment configuration.

SebastianSchildt, 2020-07-14:
Ok I made a sketch, how it might look like "in reality". I think it is not so far away from our discussion https://github.com/GENIVI/vehicle_signal_specification/pull/164 .

As a toy example lets use the one from the PR. In our VSS spec (so everything in the spec folder) I would assume a snippet like this

```yaml
- Door:
  type: instance
  #include SingleDoor.vspec
  description: All doors, including windows and switches
```

This is quite similar to the status quo: You know  - by position/include - that this is talking about `Vehicle.Cabin.Door` and you know by including SingleDoor.vspec  what kind of signals to expect, thus you have  defined

- how a "door" looks like
- where doors are located in the vehicle taxonomy

This is what VSS should do. At the same time you did not define how many doors a vehicle has (nor define an upper limit by putting 256 doors...). This is an improvement, because we want VSS to be good for big cars, small cars, trucks, busses pickups, trikes, motor bikes,....

Therefore I suggest marking the type as "instance" as an indicator: There may be one, none or more of this and  moving instantiation information to another file, e.g.`deployment.yaml`

```yaml
  - Vehicle.Cabin.Door
  instances:
    - Row[1,4]
    - ["Left","Right"]
``` 

Then stuff everything in our tooling, which would look like this:

[image]

The output after tooling has basically the same options as discussed in https://github.com/GENIVI/vehicle_signal_specification/pull/164 : Instances still visible or not? How does a "nice" json, csv look like and is this even a VSS question? 

What we probably should agree on is a "flattened" yaml (probably format not 100% right, just created it directly here...) like

```yaml
- Door:
  type: branch
  description: All doors, including windows and switches

- Door.Row1:
  type: branch
  description: Front doors

- Door.Row1.Left:
  type: branch
  description: Left front door
  - IsOpen:
    datatype: boolean
    type: actuator
    description: Is door open or closed
    ...
```

This is basically rolling in the includes and rolling out the instances (or again: Maybe keep instances "detectable")

I think it is worth doing this separation between "VSS spec" and concrete instantiation. That way we can keep the VSS data model consistent and do not need to decide for very specific instantiations. From a technical standpoint it is just seperating data into two different locations. We can provide "deployments" for a standard car as example.

Tool-wise it may be a starting point for some more general layering concepts, to enrich the data with concepts not mandatory in the core model, but let's discuss that another time :)

UlfBj, 2020-07-15:
I would propose not to assign the data type instance to the Door node, but keep it as a branch, and include a "default" instances declaration in it in the vspec file instead.
```
- Door:
  type: branch
  instances:
    - Row[1,2]
    - ["Left","Right"]
  #include SingleDoor.vspec
  description: All doors, including windows and switches
```
and then change this through overwriting from the deployment file. 
Saves us from introducing the instance data type, and having the tool start to modify node types. 
It also defines the "instance space" in the vspec file, in this case two dimensions, and provides an example of what "range" they are expected to have.

Also, in the Tool output there is shown description data in the generated instance branches, e. g. "Front doors". Where does this come from?

SebastianSchildt, 2020-07-15:
Description data in tool output: My "tool" output was approximated by hand, so yeah I would assume this information would not be there (unless somehow explicitly defined in deployment.yaml)

Actually I have also been thinking about not introducing a new node type (as I expect(ed) not so easy discussions about this), but I came to the conclusion that this is not optimal. These are my reasons:

1. I think you did the same as me, you just don't "admit" it:  Your example include instances directly in a branch in a "core" vspec file.   Now you argue, they can be overwritten in a deployment.yaml, but the question then is: What is the _meaning_ of instances in the .vspec? If it is only an "example", it is still hard to decide what a sensible example is. _If_ in .vspec it only defines a range or dimensionality, we need to clearly define what is meant by this e.g. is your example forbidding a Row3? Or a middle seat? Or "floors" in a bus? If that is all allowed, basically putting any "instance" metadata has the same meaning as just "tagging" a branch as instantiatable.  In my view is, this strengthens the point, that instances should not be part of the core model.

2. Another solution might be to have only branches, and say they are in principle instantiatable. But that does probably not make sense for all nodes. If somebody decides to instantiate "Vehicle" or "Vehicle.Cabin", because he can, it might make the data model look more surprising to applications and a robust app need to do a lot of checking how the concrete tree looks like. It takes some of the simplicity away as it can not expect "Vehicle.Speed" to always be there because Vehicle[companyA][FL][129].Speed would be totally "valid" also.

In the end, if there are arguments for it or just  a majority here wants to keep instances in vspec, I can live with that, **IF** we have a defined way for overriding this. Because I strongly believe doing these kind of adaptations to a concrete vehicle is something _every_ user of VSS needs to do, thus we should provide a standardized/canonical way to do so (while keeping it simple)

UlfBj, 2020-07-16:
>>  If it is only an "example", it is still hard to decide what a sensible example is. 

I would like to see it as follows.
Normative parts: The instantiation is tied to this node. The dimension of the instantiation.
Non-normative parts: The range of a dimension. 
So it is both an example, and not. 
To adhere to the standard, instantiation cannot be inserted in random nodes. Only the range of instantiations in the "core" vspec is then allowed to be changed. 
This eliminates the problem presented in the bullet 2 in previous comment. It is possible to do, but then it breaks the standard.
The range can be exemplified with "minimum" ranges. Should not be so hard to come up with something reasonably sensible.
The way to apply overriding from the deployment file must be well defined.

danielwilms, 2020-07-16:
> I would like to see it as follows.
Normative parts: The instantiation is tied to this node. The dimension of the instantiation.

I don't know if I'd fully agree. We had the discussions about cardinality of a branch with an uncertain number of elements, similar to the array we have on leaf node level, which could be regarded here as well.

The yaml could look like:
```yaml
Door:
  type: branch
  instantiated: true
```

and the deployment could be like (I add all alternatives I see and it's not realistic):
```yaml
Door:
  instances:
    - [...]
    - Row[1,...]
    - Row[1,2]
    - ['Left','Right']
```

The keywords like `Row`, `Left`, `Right` could be a subset of a predefined list of possible keywords to not lose the semantics behind it.

UlfBj, 2020-07-16:
If you by cardinality mean what I called range, then I see no problem. If you mean the number of dimensions of the array, I see it as problematic. 
I support the idea of a list of possible keywords.

It seems to me that you propose that the only "indication" of an insatantiation in the yaml is "instantiated: true", and then the actual instantiation declaration is in the deployment file?
I think that opens up for interoperability problems. 
Reading the yaml (=standard) gives no clue on the characteristics of the instantiation.

SebastianSchildt, 2020-07-19:
Just following this with interest. I am sort of on the fence here. I think just a marker (whether "instantiated" keyword or a node type -> both fine), is a good workable minimum solution. 

I however also agree, it would be nice to know a little more, what to expect here in terms of instantiation, but I am not sure what is a practical set of characteristics here that can be easily expressed. If we start modelling things like "There may be at least one at most 10 dimension here, which may either be numbered, with a a preferred prefix of "row" or "deck" or should be separated according to keywords "right, left, middle", where two keyword dimensions are not allowed to follow each other in the decomposition".... Well let's just agree that is out of the realm of "practical" 

@UlfBj Can you give an example what you think is workable? Are you just thinking about a list of keywords/qualifiers such as (row, left,right)?

UlfBj, 2020-07-20:
The reason behind the instantiation concept, as I understood it, was to make the YAML description of the tree simpler, easier to read. Even if it also makes it easier to realize different instantiations, I think that is a side effect that we should keep in leash. 
1. We could be very strict, and only allow the "instantiator" to change the cardinality on instantiations in the YAML. This is the level of freedom that was offered in the legacy VSS (even if not strictly enforced). 
2. We could also allow the instantiator to change the number of dimensions.
3. We could further allow the instantiator to also apply instantiations anywhere in the tree. 
I think 2. is an appropriate level of instatiator freedom. And then maybe with some further restrictions, which could be that it is only allowed to remove dimensions, not to add. 
An example could be Seat below, where three dimensions might be declared. A doubledecker bus would need all three, many vehicles the first two, and bikes only the first.
```
Seat :
  instances:
    - Row[1..2]
    - ["Left", "Right"]
    - Floor[1,2]
```
Removing dimensions could then be done in the deployment file as shown below for a motor bike with only one seat. The Tool would interpret the 2nd and 3rd dimension declaration as "null" and leave them out.
```
Seat :
  instances:
    - Row[1..1]
    - []
    - Floor[]
```
Regarding a list of key words, maybe a plain list is enough (Row[], Left, Right, Center, ...). Or two lists, to differentiate between Row[] and Left?

SebastianSchildt, 2020-07-22:
That sounds like a sane compromise/starting point to me. I guess only usage will show over time, how much flexibility vs. how predefined structure helps.

danielwilms, 2021-05-25:
Instances now in VSS implemented

closed 2021-05-25 by danielwilms