COVESA/vehicle_signal_specification #167: Merge and structure Powertrain and Drivetrain branches

dakrbo, 2020-07-14:
Hello everyone, as pre-sounded in Slack, I'd like to table a proposal:

In the current state of the VSS tree, there exists a somewhat incoherent split between Drivetrain and Powertrain,  where Powertrain contains some EV-Battery stuff (energy storage), whereas EV Motor (power source) is located in Drivetrain. At the same time, fuel system (energy storage) and combustion enginer (power source) are both located in Drivetrain. There currently is no structural distinction between energy storage and power source.

I would like to propose some structural changes to address these two topis:
1. merge Powertrain and Drivetrain branches
2. introduce structure elements to distinguish energy storage from power sources

Note: I suggest the get rid of 'Drivetrain' as a top-level element, and use 'Powertrain' as the top-level branch name. Rationale: [Drivetrain](https://en.wikipedia.org/wiki/Drivetrain) is a subset of [Powertrain](https://en.wikipedia.org/wiki/Powertrain) (power delivery vs power generation and delivery), as per the corresponding wikipedia definitions.

So I would propose the following changes:
- create Powertrain.EnergyStorage
- create Powertrain.PowerSource
- move Drivetrain.ElectricMotor to Powertrain.PowerSource.ElectricMotor
- move Drivetrain.Engine to Powertrain.Powersource.CombustionEngine
- move Drivetrain.FuelCell to Powertrain.EnergyStorage.FuelCell
- move Drivetrain.FuelSystem to Powertrain.EnergyStorage.FuelSystem 
- move Drivetrain.Transmission to Powertrain.Transmission 
- move Powertrain.Electric to Powertrain.EnergyStorage.Battery
- get rid of Drivetrain 

Note: I'm not sure what to do with Powertrain.Powertrain, and whether Powertrain.Transmission is what we want. So in addition, we might to something like:
- create Powertrain.Drivetrain
- move .Transmission to Powertrain.Drivetrain.Transmission
- move .Powertrain to Powertrain.Drivetrain.Electric 

That's it for the restructuring proposal - let me know what you think!

klotzbenjamin, 2020-07-14:
This is interesting, especially since there is some overlap on electric motor branches.

Just for understanding, the taxonomy would look like this, right?

- Powertrain
   - EnergyStorage
      - FuelCells
      - FuelSystem
      - Battery
   - PowerSource
      - ElectricMotor
      - CombustionEngine
   - Transmission
   - Drivetrain
      - Transmission
      - Electric

dakrbo, 2020-07-14:
Jepp - minus the Powertrain.Transmission node though (as it's already located under .Drivetrain)

SebastianSchildt, 2020-07-14:
Ok I am a little Puzzled what would be under "Driverain.Electric", becasue OP said 

> move .Powertrain to Powertrain.Drivetrain.Electric

However, most stuff here
https://github.com/GENIVI/vehicle_signal_specification/tree/master/spec/Powertrain
woud go to EnergyStorage.Battery right?

dakrbo, 2020-07-14:
Powertrain/Electric.vspec I would move to Powertrain/EnergyStorage/Battery.vspec
Powertrain/Powertrain.vspec I'd probably move to Powertrain/Drivetrain/Electric.vspec

I'm not enthusiastic about the second one; however this issue is about creating some unified overall structure - once we have that, follow-up tasks would be about maybe cleaning up/renaming/etc some of the odd things in the tree.

danielwilms, 2020-07-14:
> Powertrain/Powertrain.vspec I'd probably move to Powertrain/Drivetrain/Electric.vspec

It was argued, that it doesn't only concerns electric drivetrains, so it's straight under Powertrain.

For the rest it looks reasonable and I wouldn't have any objections. I'd propose to create a PR and then we see if there are some questions on the details.

gunnarx, 2020-07-14:
Generally OK with the whole proposal except 
- Although the first text says to get rid of Drivetrain, I'm not ruling out that Drivetrain might have a reason to reappear some time later.
- Agree, as others have pointed out, that *.Drivetrain.Electric seems not so useful - presumably most of the power delivery components are the same regardless of powersource
- The proposal later actually introduces Drivetrain for "Powertrain.Drivetrain.Transmission".
It feels somewhat unnecessarily long to me, so I would also be OK with Powertrain.Transmission.  Let's see what the majority opinion is.

gunnarx, 2020-07-14:
I also see no major problem with a general file named Powertrain under the directory named Powertrain, if we have signals that fit into such a file.  But if they are split into specialized files that is OK too.   Just be aware that the directory/file structure does not define the signal tree structure since the #include statements are controlling that - it is only a good convention that the directory/file structure also matches as far as it makes sense.

dakrbo, 2020-07-14:
Oh that's a good point, with the #include statements.

My proposal would include touching those as well, to mirror the directory structure in the tree. But I'll need to look into these a bit first.

gunnarx, 2020-08-04:
> move Drivetrain.ElectricMotor to Powertrain.PowerSource.ElectricMotor

I would be careful not to overdo the number of hierarchy levels.  Motors are such a fundamental part, maybe the most fundamental part of Powertrain, that I could probably be OK with the shorter:

``  Powertrain.ElectricMotor
    Powertrain.CombustionEngine
``

I would keep it in mind, here and elsewhere, to use intermediate branches primarily if the number of sibling items are so large (or so unrelated) that it is really required for organization.

gunnarx, 2020-08-04:
Looks OK to me otherwise, so a simple majority vote on the naming should be fine.

SebastianSchildt, 2020-08-07:
Can be closed since PR is merged. I can't but  @danielwilms  @gunnarx @dakrbo  should be able to

closed 2020-08-11 by gunnarx