COVESA/vehicle_signal_specification #170: High-level data and data around the vehicle

JochenKienzle, 2020-07-23:
Hello everyone,

I started to deep dive into VSS since beginning of this year. For some use cases, I would need to access high-level data and data related to the 'situation in and around the vehicle'.

On the one hand, this means data for specific entities/domains. 
Two examples for that:

1.  Temperature outside the vehicle
This value is likely to be measured by a temperature sensor 'in' the vehicle but relates to something "outside the vehicle".

2. Driver distraction
This value is strongly related to a "driver" but would be likely defined by multiple sensors (e.g. driver monitoring, loudness, number of occupants,…).

On the other hand, there could be a higher description of what is going on in and around the vehicle which is not related to a physical entity. It would describe this as the 'scene'. Such information is something which lies above or parallel to the vehicle. For example, this could be the overall "driving complexity" which is defined by multiple data (e.g. traffic density & road type) and is related to the hole vehicle.

For both types I have the following questions:

**1. Shall such data be part of the VSS?**
My proposal: Yes, because such data is meaningful for vehicle applications (e.g. driver distraction could be a critical information to increase the user experience and safety) and such data is already used (e.g. showing outside temperature, driver monitoring).
	
**2. Where to place such data points? Would it be more related to a physical sensor or an abstract/logical entity?**
First idea (sketch) with some example data points:

- Vehicle
    - Cabin
        - Driver
            - Driver Distraction
    - Environment
        - Atmospheric
            - Temperature
        - Road
            - Traffic Density
            - Road Type
    - Scene
        - Driving Complexity

klotzbenjamin, 2020-07-23:
Hi,

The purpose of VSS is indeed to model the information captured or produced by vehicle sensors/actuators, and your example of temperature outside the vehicle is relevant. This signal is already in the model as `Vehicle.AmbientAirTemperature` and relates directly to the high-level [Vehicle branch](https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Vehicle/Vehicle.vspec) like many "generic" signals (such as the IdleTime).

Concerning the use of external branches (Environment, Scene), my view is that they represent more vehicle components than use cases, and with your examples I would imagine something like:

- Vehicle
  - Cabin
    - DriverMonitoringSystem
      - Driver Distraction
    - Infotainment
      - Navigation
        - TrafficSensity
  - ADAS
    - SceneMonitoring
      - Driving Complexity

Many contextualizing sensors could be seen as part of ADAS too. 
And in the end I think such sensors would be great proposals for extensions through private branches as they relate to specific systems rather than core vehicle functions.

JochenKienzle, 2020-07-27:
Hi Benjamin,
thanks for your feedback and the proposed solution.

For now, it seems to be that this would work out for our data points.

What's about the idea of private branches? Something like an add-on-"branch" which applies for vehicles which provides the needed features? Would such branches be identical for the specific topics?

klotzbenjamin, 2020-07-27:
> What's about the idea of private branches? Something like an add-on-"branch" which applies for vehicles which provides the needed features? Would such branches be identical for the specific topics?

[Private branches](https://genivi.github.io/vehicle_signal_specification/rule_set/private_branch/) are the historical approach to extensions for VSS, to include specific technologies from a single OEM for instance. They behave similarly as regular branches.

JochenKienzle, 2020-07-27:
Thanks for clarifying. 
It looks like private branches are mainly used to store proprietary stuff but this is not my intention for this kind of data. These generic data and high-level situational data points are valid for 'all' vehicles, but could not be measured/estimated by all of them. If it is going more to a plug-in/add-on approach, then it would fit better (e.g. Driver Monitoring / Scene Monitoring / ...) and can be re-used for all vehicles which provides such features.

Especially the idea of a "Scene Monitoring" introduces a virtual device, right? It is very likely, that it uses different sensors and approaches/algorithms in a vehicle to estimate the driving complexity (as an example). So, it isn't a physical entity in the vehicle like a wheel or a seat.

danielwilms, 2020-07-28:
We have to be careful not to overload VSS. In the W3C there was a discussion once around bringing [different taxonomies together](https://github.com/w3c/automotive/issues/306). I think the point is valid still. How to treat media or a person within the vehicle context would be interesting to think about.

At the end we have to decide with each node we're adding and maybe define best practices. The broader issue would be something for the W3C group I think.

SebastianSchildt, 2020-07-28:
It is true, the "core" VSS data model should not include/require every data point you can think of, however I think your argument @danielwilms  would support the idea either not doing it at all, or clearly having defined extension points (standardised or not), such as "driver" or "environment" where such things can be modelled, instead of "sprinkling" only the most important (for any definition of important...) data all around the existing vehicle-centric tree.

Whether this should be standardised, recommended, be in this repo or another is of course another discussion. But I think these ideas confirm that on the tooling/meta layer we should have a solid layering/composition concept, where you can extend/combine the core VSS with a set of private or (semi-)standard optional data.

danielwilms, 2020-08-26:
> It is true, the "core" VSS data model should not include/require every data point you can think of, however I think your argument @danielwilms would support the idea either not doing it at all, or clearly having defined extension points (standardised or not)

With VSS "core", I meant everything under the non-private folder of the spec folder included in the Vehicle.spec file. To start such an extension in the private branch would definitely be an option. 

@JochenKienzle: maybe you want to follow as well #176

SebastianSchildt, 2020-08-26:
It may be a nitpick, that can be changed later anyway, but I would argue "Private" really is the scratchpad area where you can do whatever proprietary things you want. So yes, it is the area where "anything goes" by definition. I would however suggest once there are things that are generally useful, they should be promoted to be sorted logically, i.e. (just an example, not sure those are good concrete examples), having an `Environment`  top-level branch or a `Vehicle.Diagnostics` sub-branch . We might want to split such things later into a separate Repo/folder (layers come to mind do compose this)  and not "Mix it" with he spec folder here, to make clear what is "vehicle core" and what is an optional extension.

Anyway for trying out first things and discussing concrete structures for "high level data around the vehicle", the "Private" branch might just be the right destination for now

JochenKienzle, 2020-08-26:
@danielwilms Thank you for the hint. You are right, #176 has the same intention as this issue.
Furthermore, it has already overlapping data points (e.g. Driver Distraction). In addition to #176 something like 'Environment' would still make sense for me to complete the big picture. I'll check and would prepare a pull request for further discussions.

danielwilms, 2020-08-26:
> I would however suggest once there are things that are generally useful, they should be promoted to be sorted logically, i.e. (just an example, not sure those are good concrete examples), having an [...]

@SebastianSchildt I think this one (environment), and maybe the occupant as well, would be good concrete examples for the  layering and the separation of the different domains. Different top-level branches might be one option among others.

danielwilms, 2020-08-26:
> In addition to #176 something like 'Environment' would still make sense for me to complete the big picture. I'll check and would prepare a pull request for further discussions.

@JochenKienzle Looking forward to it!

danielwilms, 2021-06-01:
Implemented in #176, for more related signals please open PR, or reopen this issue, if a discussion is needed before that.

closed 2021-06-01 by danielwilms