COVESA/vehicle_signal_specification #180: Support ACEA Level A1 Proposed Data List

SebastianSchildt, 2020-08-25:
Industry is discussing, what kind of data a vehicle should be able to provide, and who should be able to access waht and where.

In VSS we want to create technology to support use cases from different players. [ACEA](https://www.acea.be) has a list of 42 basic data items, which should be able to be expressed in VSS, if within scope

Hre is the list for A1 level access:

[acea42Listb.pdf](https://github.com/GENIVI/vehicle_signal_specification/files/5123874/acea42Listb.pdf)

For reference, this is ACEAs view on data different tiers of data access

- **Level A** data in the vehicle is available on the OEM backend via the Extended Vehicle web services interface. This can be accessed by a 3rd party either via an OEM server or via a Neutral Server. The customer has provided consent to the OEM for the transfer of level A data from their vehicle to the OEM server. Further consent is required from the customer for the OEM to transfer level A data from the OEM server to their chosen 3rd party.
-  **Level A** data is partitioned into a predefined data set **(A1)** and additional data used by an OEM **(A2)** for the generation of their own remote services. The predefined data set is data provided by all vehicle manufacturers provided that the vehicle in question is equipped to deliver the data and the data is available on the OEM server. Level A data can be accessed at the sampling frequency implemented and used by the OEM.
- **Level B** data is data available on a vehicle interface but has not yet been transferred to an OEM server. OEMs can only transfer data for specific purposes, with the consent of the customer. For a third party to access level B data, a B2B agreement will be required with the OEM. The data will be transferred specifically for their usage. The data can then be channeled to the third party via the OEM web interface or a Neutral Server, provided the vehicle owner has provided consent for the data transfer.
- **Level C** data is data that is generated by routines hosted in a vehicle’s electronic control unit (ECU). The routines generated by third parties can be used to aggregate and process data on-board. A B2B contract will be required between the OEM and third parties wishing to develop, integrate, test and deploy routines. Access to vehicle resources and the transfer of data generated by third party routines will be governed within the B2B contract.

klotzbenjamin, 2020-08-27:
Regarding the A1 list you provided, I quickly looked into what VSS can describe:

1. Vehicle.Speed
2. Vehicle.Acceleration.Longitudinal
3. Vehicle.TravelledDistance
4. Multiple OBD signals referring to DTC (e.g. Vehicle.OBD.Status.DTCCount), but not DTC branch
5. Vehicle.Powertrain.BatteryManagement.Battery.StateOfCharge or Vehicle.Powertrain.Electric.Battery.StateOfCharge
6. Vehicle.Cabin.Infotainment.Navigation.CurrentLocation.Latitude/Longitude (VSS does not provide a Time signal)
7. ?
8. ?
9. ?
10. Vehicle.Drivetrain.FuelSystem.Range ? (Do we have a range for electric vehicles?)
11. Vehicle.Drivetrain.FuelSystem.Level
12. Vehicle.Drivetrain.FuelSystem.Range
13. Vehicle.Chassis.Axle.RowX.Wheel.Left/Right.Brake.PadWear
14. Vehicle.Powertrain.Electric.Battery.StateOfCharge? (or do they mean a boolean like CurrentlyCharging)
15. Vehicle.Drivetrain.FuelSystem.LowFuelLevel
16. Vehicle.Cabin.Seat.RowX.PosX.IsBelted
17. ?
18. ?
19. No Power steering signals
20. Vehicle.Cabin.Door.RowX.Left/Right.IsLocked
21. No signal for the warning lamp
22. No signal for the warning lamp
23. No signal for the warning lamp
24. No signal for the warning lamp
25. ?
26. Vehicle.Chassis.Axle.Row1.Wheel.Left.Brake.FluidLevelLow
27. Vehicle.Cabin.Door.IsLocked? (it depends on the position interpretation)
28. ?
29. ?
30. ?
31. Vehicle.OBD.FuelPressure
32. ?
33. Vehicle.Chassis.Axle.RowX.Wheel.Left/Right.Tire.PressureLow
34. No warning?
35. Vehicle.Body.Windshield.Front.WasherFluid.Level
36. Vehicle.Cabin.Door.Row1.Left.Window.Position/IsOpen
37. Vehicle.Cabin.Sunroof.Position
38. Vehicle.Cabin.Door.RowX.Left/Right.IsOpen (and more signals in the Door branch)
39. Vehicle.Body.Trunk.IsOpen
40. Vehicle.Drivetrain.BatteryManagement.LowBatteryLevel or Vehicle.Powertrain.BatteryManagement.Battery.LowLevel (no warning lamp signal)
41. ?
42. ?

Beyond the signals and the data they convey, they seem to focus more on the warning signs rather than their causes, and the question of what unit/format remains fully open.

SebastianSchildt, 2020-08-27:
Hi Benjamin, thanks for taking a look. Yes, a lot is already there. And some stuff is not a good fit for VSS directly, I already discussed with @danielwilms offline that all those "XXXWarningLamp" are not very helpful form the semantic poit of view. I like VSS to tell me teh oil pressure, probably I also want it to tell me when it is Critical/Low. But telling me whether a Specific LED is on in my Dashboard is of limited value.

NItpick: 31 "Oil" != "Fuel" 

I just want to use this issue to track progress so thanks for starting a list 👍

SebastianSchildt, 2020-10-13:
***Tracking status***

| Done | A1 Item | VSS Path | Comment | 
|-- | --- | --- | --- |
|:white_check_mark: |  1. Speed of Vehicle | Vehicle.Speed | - |
|:white_check_mark:| 2. Acceleration of Vehicle | Vehicle.Acceleration | - |
|:white_check_mark:| 3. Odometer Value | Vehicle.Powertrain.Transmission.TravelledDistance | - |
|:construction: | 4. Diagnostic Trouble Codes | TBD | https://github.com/GENIVI/vehicle_signal_specification/pull/177 |
|:white_check_mark:| 5. Vehicle High Voltage Battery State of Charge | Vehicle.Powertrain.Battery.Battery.StateOfCharge | -|
|:construction:| 6. GPS position and Time at Current Value | Vehicle.Cabin.Children.Infotainment.Navigation.CurrentLocation | what means "Time"? |
|:question: | 7. Service Indicator | ? | What is this? MIL? |
|:x: | 8. Distance to next service | TBD | - |
|:x: | 9. Time to next service | TBD | - |
|:question: | 10. Remaining Electric Range/Distance | Vehicle.Powertrain.FuelSystem.Range | Difference to 12?  |
|:white_check_mark: | 11. Fuel Tank Level | Vehicle.Powertrain.Fuelsystem.Level | - |
|:question: | 12. Total Remaining Range | Vehicle.Powertrain.FuelSystem.Range | Difference to 10 |
|:white_check_mark: | 13. Brake Pad Low Warning | Vehicle.Chassis.Axis.Wheel.Brake.PadWear/BrakesWorn | - |
| :x:| 14. High Voltage Battery Charging Status | TBD | Probably should be part of Battery or Charging Port |
|:x: | 15. Low Oil Indicator |  TBD | |
|:white_check_mark: | 16. Buckles Status | Vehicle.Cabin.Seat.isBelted | - |
|:x: | 17. Diesel Additive Range  | TBD | - |
|:x: | 18. Diesel Additive Status (Low Indicator)  | TBD | - |
|:x: | 19. Power Steering Warning Lamp | TBD | ignore "lamp" |
|:white_check_mark: | 20. Door Lock Status | Vehicle.Cabin.Door.isLocked | - |
|:x: | 21. Airbag Warning Lamp | TBD | ignore "lamp" |
|:x: | 22. Coolant Temperature High Warning | TBD | We have Vehicle.Powertrain.ElectricMotor.Motor.CoolantTemperature Vehicle.OBD.CoolantTemperature and Vehicle.Powertrain.CombustionEngine.Engine.ECT |
|:question:| 23. Malfunction Indicator Status | TBD | Is it MIL? |
|:question: | 24. Standby/Quiescent Current Status | TBD | Is it Vampire drain? |
|:x: | 25. Distance until Engine Oil Service | TBD - |
|:white_check_mark: | 26. Brake Fluid Level Low Warning | Vehicle.Chassis.Axis.Wheel.Brake.FluidLevelLow | - |
|:question: | 27. Full Lock Status | What is this? |
|:x: | 28. Remaining Brake Fluid Life | TBD | - |
|:x: | 29. Distance until Brake Service | TBD | -|
|:x: | 30. Due Date for Brake Fluid Change | TBD | What is the difference to 28? |
|:x: | 31. Oil Pressure Warning Lamp | TBD | we habe Vehicle.Powertrain.CombustionEngine.Engine.EOP |
|:x: | 32. Time until next Required Oil Change | TBD | - |
|:white_check_mark: | 33. Tyre pressure warning | Vehcile.Chassis.Axis.Wheel.Tire.PressureLow | - |
|:x: | 34. Oil Temperature High Warning | TBD | - | 
|:x: | 35. Windshield Wiper Fluid Level | TBD |- |
|:white_check_mark: | 36. Window Status | Vehicle.Cabin.Door.Window.Position , Vehicle.Cabin.Door.Window.isClosed | - |
|:white_check_mark: | 37. Sunroof Status | Vehicle.Cabin.Sunroof.Position | see 42 |
|:white_check_mark: | 38. Door Status | Vehicle.Cabin.Door.isOpen, Vehicle.Cabin.Door.isLocked | - |
|:white_check_mark: | 39. Trunk Status | Vehicle.Body.Trunk.isOpen, Vehicle.Body.Trunk.isLocked | -| 
|:x: | 40. Battery Lamp Status | TBD | Ignore "lamp" |
|:x: | 41. Convertible Roof status | TBD | - |
|:white_check_mark: | 42. Tilting Status of Sunroof | Vehicle.Cabin.Sunroof.Position | see 37 |

I will try to keep this updated

chheis, 2020-10-13:
7. Service Indicator = Service needed (Wartungsintervall erreicht https://www.yourmechanic.com/article/all-about-car-service-indicator-lights-by-brady-klopfer)
Same is with 8. + 9. (just the distance to the next needed service)
12. in case of PHEV you need 2 different range indicators (one for combustion and one for the EV) IMHO shall be within Powertrain directly
14. ==> Powertrain.Battery.Charging.Status
15. ==> CombustionEngine.Engine.EOP        ==> Maybe here?? (in the Motorbike it is the same)
23. Yes ==> MIL
24. Current with Vehicle off or in standby (remote services are available etc.)
25. ==> Same as 8.?
27. Normal lock = no Alarm sensors on / Full lock = all alarm sensors on (e.g. tilt sensor / cabin sensor etc)
34. ==> CombustionEngine.Engine.EOT

SebastianSchildt, 2021-12-07:
***Tracking status***

*Update for VSS 2.2* 

| Done | A1 Item | VSS Path | Comment | 
|-- | --- | --- | --- |
|:white_check_mark: |  1. Speed of Vehicle | Vehicle.Speed | - |
|:white_check_mark:| 2. Acceleration of Vehicle | Vehicle.Acceleration | - |
|:white_check_mark:| 3. Odometer Value | Vehicle.TravelledDistance | - |
|:white_check_mark: | 4. Diagnostic Trouble Codes | Vehicle.OBD.DTCList |  |
|:white_check_mark:| 5. Vehicle High Voltage Battery State of Charge | Vehicle.Powertrain.Battery.Battery.StateOfCharge | -|
|:construction:| 6. GPS position and Time at Current Value | Vehicle.Cabin.Children.Infotainment.Navigation.CurrentLocation | Time https://github.com/COVESA/vehicle_signal_specification/pull/379 |
|:white_check_mark: | 7. Service Indicator | Vehicle.Service.ServiceDue |  |
|:white_check_mark: | 8. Distance to next service | Vehicle.Service.DistanceToService | - |
|:white_check_mark: | 9. Time to next service | Vehicle.Service.TimeToService | - |
|:white_check_mark: | 10. Remaining Electric Range/Distance | Vehicle.Powertrain.Battery.Range | Remaining range in meters using only battery.|
|:white_check_mark: | 11. Fuel Tank Level | Vehicle.Powertrain.Fuelsystem.Level | - |
|:white_check_mark: | 12. Total Remaining Range | Vehicle.Powertrain.Range |  Remaining range in meters using all energy sources available in the vehicle.|
|:white_check_mark: | 13. Brake Pad Low Warning | Vehicle.Chassis.Axis.Wheel.Brake.PadWear/BrakesWorn | - |
| :white_check_mark: | 14. High Voltage Battery Charging Status | Vehicle.Powertrain.Charging.* | i.e. ChargeCurrent/ChargePlugStatus, ...  |
|:white_check_mark: | 15. Low Oil Indicator |  Vehicle:powertrain.CombustionEngine.EngineOilLevel | states "low" and "critically low"|
|:white_check_mark: | 16. Buckles Status | Vehicle.Cabin.Seat.isBelted | - |
|:x:  | 17. Diesel Additive Range  | TBD | - |
|:x: | 18. Diesel Additive Status (Low Indicator)  | TBD | - |
|:x: | 19. Power Steering Warning Lamp | TBD | "lamps" would not be modelled, but currently no power steering at all |
|:white_check_mark: | 20. Door Lock Status | Vehicle.Cabin.Door.isLocked | - |
|:x: | 21. Airbag Warning Lamp | TBD | "lamps" would not be modelled, but currently no airbag system state, only "IsDeployed" |
|:x: | 22. Coolant Temperature High Warning | TBD | We have Vehicle.Powertrain.ElectricMotor.Motor.CoolantTemperature Vehicle.OBD.CoolantTemperature and Vehicle.Powertrain.CombustionEngine.Engine.ECT but no warning |
|:white_check_mark:| 23. Malfunction Indicator Status | Vehicle.OBD.Status.MIL |  |
|:x: | 24. Standby/Quiescent Current Status | TBD |  Vampire drain? |
|:?: | 25. Distance until Engine Oil Service | TBD | Have Vehicle.Powertrain.CombusionEngine.OilLifeRemaining in _seconds_ and Vehicle.Service.DistanceToService |
|:white_check_mark: | 26. Brake Fluid Level Low Warning | Vehicle.Chassis.Axis.Wheel.Brake.FluidLevelLow | - |
|:x: | 27. Full Lock Status |  | isLocked for all doors and trunk seperately|
|:question: | 28. Remaining Brake Fluid Life | TBD | only Vehicle.Chassis.Axle.RowX.Wheel.Left/Right.FluidLevel (%) and general Vehicle.Service.TimeToService |
|:question: | 29. Distance until Brake Service | TBD | only Vehicle.Chassis.Axle.RowX.Wheel.Left/Right.FluidLevel (%), PadWear (13) and general Vehicle.Service.DistanceToService |
|:x: | 30. Due Date for Brake Fluid Change | TBD | What is the difference to 28? |
|:x: | 31. Oil Pressure Warning Lamp | TBD | "lamps" not mocelled in, we have Vehicle.Powertrain.CombustionEngine.Engine.EOP |
|:x: | 32. Time until next Required Oil Change | TBD | only Vehicle.Serivce.TimeToService |
|:white_check_mark: | 33. Tyre pressure warning | Vehcile.Chassis.Axis.Wheel.Tire.PressureLow | - |
|:x: | 34. Oil Temperature High Warning | TBD | Vehicle.OBD:OilTemperature, but no high aring| 
|:white_check_mark: | 35. Windshield Wiper Fluid Level | Vehicle.Body.Windhield.Front.WasherFluid.LevelLow | .Level in % also available |
|:white_check_mark: | 36. Window Status | Vehicle.Cabin.Door.Window.Position , Vehicle.Cabin.Door.Window.isClosed | - |
|:white_check_mark: | 37. Sunroof Status | Vehicle.Cabin.Sunroof.Position | see 42 |
|:white_check_mark: | 38. Door Status | Vehicle.Cabin.Door.isOpen, Vehicle.Cabin.Door.isLocked | - |
|:white_check_mark: | 39. Trunk Status | Vehicle.Body.Trunk.isOpen, Vehicle.Body.Trunk.isLocked | -| 
|:x: | 40. Battery Lamp Status | TBD | Ignore "lamp", assuming LV system, we only habe Vehicle.OBD.ControlModuleVoltage |
|:question: | 41. Convertible Roof status | TBD | Maybe Vehicle.Cabin.Sunroof.Position  |
|:white_check_mark: | 42. Tilting Status of Sunroof | Vehicle.Cabin.Sunroof.Position | see 37 |

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch