COVESA/vehicle_signal_specification #183: array datatype declaration should not contain allocation data

UlfBj, 2020-08-27:
In the documentation it says:
# undefined number of elements in an array
datatype: UInt32[]
# fixed number of elements in an array
datatype: UInt32[100]

I started to think about implementing support for arrays in the W3C Gen2 server implementation, which led me to think about the two datatype declarations above. 
I do not think it is correct to include "allocation" data in the datatype declaration, it should only declare that it is an array, and not a scalar value. The size of the array should not be declared here. If it is seen as important to have it included in the node metadata, then it could be part of the Description text. 
In the case of VSS usage in W3C automotive spec, the array data is contained in a JSON payload, where the array size is available implicitly through len(array). So there is no need to necessarily declare it in the VSS metadata, more than for the convenience of the client implementer. 
I think therefore that array datatype declarations never contain array size, thus always be the two square brackets only "[]" (following one of the scalar data types).

danielwilms, 2020-08-27:
> So there is no need to necessarily declare it in the VSS metadata, more than for the convenience of the client implementer.

It's a way to optimize the implementation, and I think a limit can be useful there, if known. It's a similar discussion to #163, where the proposal is to use min/max.

But what I didn't get from the issue, where we are running into problems with the current solution.

UlfBj, 2020-08-27:
>> It's a way to optimize the implementation, and I think a limit can be useful there, if known

It can be useful, I agree. But it can then be provided in the description instead. 
Or in a separate property: "arraysize": X

>> But what I didn't get from the issue, where we are running into problems with the current solution.

In my view it is formally not so elegant to include this information in something named "datatype". 
"string[x]" is then formally a different datatype compared to "string[y]".
Implementations will have to start parsing datatypes to separate datatype from allocation needs.

SebastianSchildt, 2020-08-28:
I am on the fence here. Your argument is true for highlevel languages, however in most/many languages string[x] is quite different from string[9] (often statically allocated with immutable sizes). 

I also think  #163 applies....

I am fine with everything :), but I agree the approach of having (max)size as a separate property would be in line with #163  having just a generic "Int"/"number" type and putting limitations/size guarantees in extra properties...

UlfBj, 2020-08-28:
>> having just a generic "Int"/"number" type and putting limitations/size guarantees in extra properties...

Maybe like this:
An extra property, here called "arraysize", could then support the following
- If not declared in the node metadata then the data is a scalar of the datatype defined.
- If present in the node metadata as "arraysize": x then the data is an array:
   For x = 0 the array size is unbounded
   For x > 0 the array size is set to x

danielwilms, 2020-09-02:
From the discussion yesterday, I second the proposal of @UlfBj, that it's rather a question of deployment than of the data model itself. That being said, I would expect it as additional metadata, like you proposed.

As for the argument of simplicity of the implementation, in my opinion it depends on the protocol definition and not of the data model. I could imagine the protocol making the `arraysize` argument mandatory for array data types, then the initialisation in the implementation can be done for the right size and not for some best guess done during specification. From my perspective, that would be easy, lean and flexible. For creating the generic data structure I think that it's enough information to know, that it's an array. And this is what the specification should focus on.

SebastianSchildt, 2020-09-08:
After thinking about this I feel @UlfBj  is the cleanest solutions: 

 - It allows us to have variable and fixed size arrays. 
 - We can set fixed size in the VSS model//catalog by setting arraysize in vspe files
 - By leaving it out, we have a variable sized array, that may be limited by adding an arraysize limitation in deployment, if we know it for a current implementation, or the used technology requires it.

danielwilms, 2020-09-15:
Decision:
```
datatype: type[]
arraysize: x
```

Document and fix where it's different.

PR #189 (2020-10-13): Changed documentation according to #183
Array is described as:

datatype: type[]
arraysize: x

Fixes: #183
closed 2020-10-13 by danielwilms