COVESA/vehicle_signal_specification #195: Include directives Relative or Absolute?

zubairhamed, 2020-09-22:
Are the #include directives relative or absolute path-ed?

In the example of Cabin/Occupant.vspec, it seems to be relative to the root (VehicleSignalSpecification.vspec)
https://github.com/GENIVI/vehicle_signal_specification/blob/339fd776d8ca5a4e569c4af024669a4c29ff8b5c/spec/Cabin/Occupant.vspec#L14

However, for Body/Body.vspec, the inclusion is relative to Body.vspec
https://github.com/GENIVI/vehicle_signal_specification/blob/339fd776d8ca5a4e569c4af024669a4c29ff8b5c/spec/Body/Body.vspec#L137

danielwilms, 2020-10-10:
@zubairhamed [documentation](https://genivi.github.io/vehicle_signal_specification/rule_set/includes/) says:

> The <filename> part specifies the path, relative to the **file with the #include directive**, to the vspec file to replace the directive with.

So the latter should be the right one according to the documentation. But should be consistent in the spec. Good find.

danielwilms, 2021-01-19:
Both are supported, which one first has to be found out. @magnusfeuer 
 Best practices needed.

Try out & write best practices @SebastianSchildt

SebastianSchildt, 2021-01-26:
Both, absolute and relative paths have merit, I suggest the following

Paths starting with slash `/` are absolute with regard to the vspec, i.e. they are relative to the root vspec file

i.e.
```
#include /Body/ExteriorLights.vspec Lights 
```

would include this file https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Body/ExteriorLights.vspec
in case the tooling has been invoked with the our standard top level vspec https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/VehicleSignalSpecification.vspec

accordingly, include paths _without_ a starting `/`are relative _to the .vspec that contains the include statement_, i.e. to include the `ExteriorLights.vspec`file from the example

in https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Body/Body.vspec do

```
 #include ExteriorLights.vspec Lights 
```

To potentially include it from https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Powertrain/CombustionEngine.vspec 

```
 #include ../Body/ExteriorLights.vspec Lights 
```

including it directly in https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/VehicleSignalSpecification.vspec

```
 #include Body/ExteriorLights.vspec Lights 
```

By basing the semantics on the filenames, and not on the "tree structure", this is robust no matter how you nest it.

I advise against file-system absolute paths (bad design, and a potential security risk). Our "root" should be the folder of the main Vspec file.  Also there are symlinks if you really have some files _above_ the main vspec

(speaking of security risk, we _might_ want to avoid relative traversal in the tools to go above the main.vspec file, but that is a tool-quality question, not a specification one)

If we agree steps would be: Modifiy tools to follow agreed search patterns, and fix the issues popping up

magnusfeuer, 2021-01-26:
Looked into the code.

tl;dr: Local directory is searched 
It is actually up to the calling program (`vspec2c.py`, `vscpec2csv.py`, etc) to determine the include path search order.

Digging into `vspec2csv.py`, we see that it starts by adding the current directory to its path, followed by any paths provided with the `-I` directive. [Link](https://github.com/GENIVI/vss-tools/blob/dcb22fd28af3f89a97b25d854d174e8f94218cd7/vspec2csv.py#L78) [Link](https://github.com/GENIVI/vss-tools/blob/dcb22fd28af3f89a97b25d854d174e8f94218cd7/vspec2csv.py#L81)

More rules:
If the include file name starts with a `/`, then it is opened as-is with no include directory search. [Link](https://github.com/GENIVI/vss-tools/blob/dcb22fd28af3f89a97b25d854d174e8f94218cd7/vspec.py#L185)

If an include file cannot be found in the provided include paths (`.` and `-I` directives), vscgen.py will do a final search in the directory hosting the file that has the `#include` directive in it. This allows VSS files stored in the same directory to refer to each other. [Link](https://github.com/GENIVI/vss-tools/blob/dcb22fd28af3f89a97b25d854d174e8f94218cd7/vspec.py#L345)

erikbosch, 2024-02-12:
Think this one can be closed

closed 2024-02-12 by erikbosch