COVESA/vehicle_signal_specification #197: Sensors vs. Actuators

danielwilms, 2020-09-30:
There is an issue with the current understanding of sensor vs. actuator.

@gunnarx argued, that contrary to the current specification, any data entry can be assumed writeable, unless it's an attribute (const). Given example: If an ECU A with a specific sensor wants to distribute its own value, it might want to do that over the one vss-enabled protocol, which the other entities are using to consume that value. In that case ECU A needs the rights to write that value and not being limited by the specification.

Some thoughts on that:
1. With the model we target primarily creating a common vocabulary within the industry and IMO even more one for non-domain experts. I personally think the fact, that there is an indication of whether a data entry is writable or not adds a lot of value, especially to the latter. 
1. The model describes concepts, which are used in the vehicle domain. And this determines, if the value can be changed or not. That a sensor has to communicate the data somehow, doesn't change that fact. If you use the argument above, there wouldn't be any place for attributes either. The additional information serves its purpose describing the concept. 
2. This was one of the first big changes towards v2. It has been discussed and is now in place. We can go forward to have the discussion for v3, and I think it fits into the scope of deployment/layering. But let's not reopen this issue and try to move on for now.

What do you think, @UlfBj, @chheis, @SebastianSchildt, @gunnarx, @adobekan?

UlfBj, 2020-09-30:
Whether a signal is read-only ,or read/write is key information for how a client will interact with it. To define this follows a well-known pattern, it is not something specific to VSS.

danielwilms, 2020-10-06:
@danielwilms make proposal how it can be written.

PR #199 (2020-10-20): Clarify description of attribute, sensor and actuator
Fixes: #197
commit 850cbe1c (2020-10-20): Clarify description of attribute, sensor and actuator

Fixes: #197
closed 2020-10-20 by danielwilms