COVESA/vehicle_signal_specification #198: Inconsistency with attribute value vs. default

rfrowe, 2020-10-08:
The documentation states:
> An attribute is a signal with a default value, specified by its `value` member.

However, the example immediately below is:
```yaml
- MaxPower:
  type: attribute
  datatype:  uint16
  default: 0
  description: Peak power, in kilowatts, that engine can generate.
```

This example is noticeably lacking a `value` and instead a `default` has been specified. Looking at the specification itself, both `default` and `value` seem to be used interchangeably in different attributes to mean the same thing. 

For example:
```yaml
# From spec/Powertrain/EnergyStorage/FuelSystem.vspec
- HybridType:
  datatype: string
  type: attribute
  enum: [ "unknown", "not_applicable", "stop_start", "belt_ISG", "CIMG", "PHEV" ]
  default: "unknown"
  description: Defines the hybrid type of the vehicle
```

```yaml
# From spec/Chassis/Chassis.vspec
- CurbWeight:
  datatype: uint16
  type: attribute
  value: 0
  unit: kg
  description: Vehicle curb weight, in kg, including all liquids and full tank of fuel, but no cargo or passengers.
```

As such, the specification is inconsistent. All should use either `default` or `value`.

danielwilms, 2020-10-10:
Agreed @rfrowe. In the documentation it's the `value`, stating:

> An attribute is a signal with a default value, specified by its value member.

Looking at it, I think `default` is a better fit. I'll create a PR to fix it.

rfrowe, 2020-10-12:
Great, thanks for simplifying my parsing logic :)

SebastianSchildt, 2020-10-13:
Should we @Floix also change the tooling, so, that a "value" in attribute is flagged as error? The point is, whether it is `default`or `value` is not that important, but inconsistency really sucks. Maybe little by little we can make our own tooling "stricter" to catch these things.

SebastianSchildt, 2021-02-10:
This is still broken in current master, e.g. 8 instances in https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Vehicle/Vehicle.vspec

Can somebody reopen @danielwilms  @rfrowe

danielwilms, 2021-02-10:
still value attributes in Vehicle and Powertrain

gunnarx, 2021-02-23:
> still value attributes in Vehicle and Powertrain

For completeness, Daniel fixed it in 67ecea313f1eec5bc54cf0d9b886d2b6ea5f80a3. 
No more value:  exists in current master.

erikbosch, 2021-03-18:
@danielwilms @gunnarx so this one can be closed?

( I think it would be good if we could do a cleanup of issues, there are quite many that are obsolete or already fixed I think)

danielwilms, 2021-05-25:
should be fixed, reopen otherwise

PR #200 (2020-10-20): Remove `value`/`default` inconsistency
In the spec value and default were used exchangeably.
Used `default` throughout the spec-files.

Fixes: #198
commit e71db4c6 (2020-10-20): Remove `value`/`default` inconsistency

In the spec value and default were used exchangeably.
Used `default` throughout the spec-files.

Fixes: #198
closed 2021-05-25 by danielwilms