COVESA/vehicle_signal_specification #216: Deprecation

danielwilms, 2020-11-10:
Some nodes are moved with in the tree. Deprecation would be help to adapt. Therefore, the suggestion is to add a `deprecation` keyword to the nodes, which are moved. Using a keyword in the yaml node works nicely with the current tooling.

Example:
```
Navigation.CurrentLocation:
  type: branch
  aggregate: true
  description: The current latitude and longitude of the vehicle.
  deprication: @v2.1 moved to Vehicle.CurrentLocation
```

SebastianSchildt, 2020-11-10:
`depr_e_cation` 😁 

Semantic of @: Does it mean moved in 2.1?, so IN 2.1 deprecated

A general question is, do we want more semantic, like "machine readable" this moved to ... or jsut a general hum readable comment? I think human readabe string is enoug (can be put out by the compiler), becasue otherwise we ran into trouble if it is "moved _and_ renamed this and that...."

Just a marker and humnan readable string is also, how it is done C++, see jsoncons as a good example (search for JSONCONS_DEPRECATED_MSG here https://github.com/danielaparker/jsoncons), basically the compiler throws those stirngs into your face as warnings whenever you touch something old.

So tldr; I think this is the approach we want, in case I read correctly between the lines. I suggest putting this into the documentation in a more verbose way, and discuss on a PR

danielwilms, 2020-11-17:
@danielwilms add a PR with the documentation.

chheis, 2020-12-11:
First real example in PR here: https://github.com/GENIVI/vehicle_signal_specification/pull/224

danielwilms, 2020-12-15:
add description for commit

PR #223 (2021-01-26): Add deprecation documentation
Fixes: #216
closed 2021-01-26 by danielwilms