COVESA/vehicle_signal_specification #219: Create SpecRef entry for VSS

tguild, 2020-11-20:
While working on VISS and bumping into a new quirk with ReSpec tool trying to get VSS as a normative reference via localBiblio mechanism, it occurred to me we should get VSS into SpecRef. SpecRef is a catalog of specification references. 

Current draft of VISS containing VSS reference.

https://w3c.github.io/automotive/vehicle_data/vehicle_information_service.html

https://www.specref.org/?q=VSS turns up nothing. I'm happy to make the entry when we agree on what it should contain.  We can make generic (eg latest) version of VSS and/or specific versions. I am inclined toward simply the former unless we want to make a new entry each time  as part of our release process. Take a look at the other potential fields, some make more sense if we have an entry per version. We may want to enumerate the various editors although if we include all contributors (12) it gets pretty expansive so may want to base on number/size of commits.

https://github.com/tobie/specref#updating--adding-new-references

https://github.com/GENIVI/vehicle_signal_specification/graphs/contributors

The JSON I have for VSS in ReSpec localBiblio[graphy]:

            VSS: {
              title: "Vehicle Signal Specification",
              href: "https://github.com/GENIVI/vehicle_signal_specification",
              //href: "https://github.com/GENIVI/vehicle_signal_specification/tree/v1.0",
              status: "1.0",
              date: "March 2017",
              publisher: "GENIVI Alliance",
            }

If we go with a single reference, would drop status and use uncommented link.

danielwilms, 2021-01-19:
general/unversioned

tguild, 2021-02-10:
Pull request made, provided 2.0 as status to reflect recent release

https://github.com/tobie/specref/pull/636

tguild, 2021-02-15:
Pull request accepted

https://www.specref.org/?q=VSS

closed 2021-02-15 by tguild