COVESA/vehicle_signal_specification #226: Trailer branch[es]

tguild, 2020-12-16:
While discussing PR 224 I wondered if Trailer should be a branch under Vehicle.

https://github.com/GENIVI/vehicle_signal_specification/pull/224

I wonder if it should remain separate for a number of reasons. 

* It is possible (physically and legally) to haul multiple trailers, two not uncommon and in some places 3 or more.

https://en.wikipedia.org/wiki/Road_train

* trailers are often decoupled and attached to different vehicles or left unattached for periods of time, as such they should be able to stand on their own as independent entities. 

* some are dynamic in nature, modular such as shipping containers attached to flatbed trailers. We will want at least an identified such as Decentralized ID (DID) for asset tracking for dumb containers, perhaps more if dynamic information available if its a refrigerated container.

https://www.w3.org/TR/did-core/

In short, I think we should take a step back and think about the different use cases around trailers so we handle them well in VSS.

danielwilms, 2021-01-19:
Model trailer as vehicle (corner case: total length) or as a branch in the specification?

erikbosch, 2021-01-20:
Is the first step maybe to discuss how we in general can or want to handle references in the model. If we for example look at the road-train example, one could consider having Next/Prev-references.  Maybe it should even be more generic so we allow references to undefined objects, or arrays of objects - could be useful for agriculture and trucks where you can mount/load various equipment. I.e. you can get a reference to the "cargo", and then you might use some discovery mechanism to find out what type of cargo it is and what data/services it offers. It might be a vehicle defined by VSS, but it can as well be something totally different. 

```
- Next:
  type: vehicle
  description: Reference to next vehicle in train, if any
- Prev:
  type: vehicle
  description: Reference to previous vehicle in train, if any
- Cargo[]:
  type: object
  description: Object reference to cargo, if any, can be of any type
```

UlfBj, 2021-04-20:
An alternative solution is found in the following presentation.
[https://docs.google.com/presentation/d/1ajL21Se50w1BKYnkIOVo5o4zTcA9PTTA9qTmFlXdSz0/edit#slide=id.g4f9a252ef2_1_94](url)

closed 2024-02-27 by erikbosch