COVESA/vehicle_signal_specification #228: Vehicle.Speed to float without limitation

chheis, 2021-01-20:
Due to calculation loses (some function may use m/s internally). Would be better to have more accuracy for the speed.
https://github.com/GENIVI/vehicle_signal_specification/blob/master/spec/Vehicle/Vehicle.vspec

erikbosch, 2021-01-21:
Do we have any general approach on when to use float/double and when to use integer types? We currently have a large number of datapoints where we use integer-representation, where someone maybe wants to have more precision in the future.  Check e.g. all temperature datapoints where it seems to be quite random whether we use integer or float.  

By the way, OBD.Speed already use float and without the -250/+250 limits

danielwilms, 2021-01-26:
1. Change speed to float. 
2. @chheis create issue number vs int/float
3. @erikbosch list of questionable integer

erikbosch, 2021-02-02:
I have identified some units (e.g. cm, l, km/h, celsius, inch, degrees/s, m/s^2) where int as I see it isn't sufficient. But float (or in rare cases double) can as I see it be considered for all datapoints where a unit is specified.

Suggestion: If we consider it reasonable to change at least the units above to float I could create separate PRs for each of them. In some cases I suggest that units are changed, but that could be discussed as part of the PRs

commit e18364d2 (2021-02-16): Using float for speed and acceleration (Fixes #228)

This makes it possible to use data with higher accuracy.

Also marking two datapoints as deprecated:
-Transmission.speed, as it is identical to Vehicle.speed
-Vehicle.accelerationTime - too ambiguous to be an attribute
closed 2021-02-16 by danielwilms