COVESA/vehicle_signal_specification #231: Improper kPa and Pa units in release 2.0

rfrowe, 2021-01-28:
As per #206 and #217, all instances of `kpa` and `pa` have been changed to `kPa` and `Pa` respectively. This change was merged well before the 2.0 release of VSS.

However, as I look at the officially released 2.0 JSON [here](https://genivi.github.io/vehicle_signal_specification/releases/v2.0/vss_release_2.0.json), I see no instances of `kPa` and `Pa` and only instances of `kpa` and `pa`. My VSS tooling depends on the correct casing of these units and this release represents a regression that breaks my tooling.

gunnarx, 2021-02-02:
I found one instance of pa, and a few instances of UInt8 and UInt32 in the spec directory, so a few instances seem to have be missed.  Our build checks were also not running all build variants so for some reason it seems to have slipped through.  Some (hopefully all) of those fixes will be soon merged because I'm working to update and merge #220 

Related to this I however question if JSON should be "released", since it is a generated artifact from the original source code.

danielwilms, 2021-02-02:
handled in #232

PR #235 (2021-02-16): Updated release files
Updated release branch as discussed in #232 and generated the new files. Those are added to the documentation.

Fixes: #231
closed 2021-02-16 by danielwilms