COVESA/vehicle_signal_specification #232: Content of v2.0, maintenance branch (and branch strategy in the future)

gunnarx, 2021-02-02:
**1) Does something in this need to be adjusted?**

Commits in **origin/master** but not in **origin/release/2.0**:

* a95f464  danielwilms  7 days ago (authored 7 days ago)  Release 2.0 (#227)
* 891eeed  danielwilms  7 days ago (authored 7 days ago)  Add deprecation documentation (#223)
* a3aeaf3  Christian Heissenberger  2 weeks ago (authored 2 weeks ago)  "@" is a reseved symbol. possible bugfix, as gunnarx proposed
* c3c2e56  Christian Heissenberger  2 weeks ago (authored 7 weeks ago)  bugfixed commit for deprecation
* ef683c4  Christian Heissenberger  2 weeks ago (authored 8 weeks ago)  promoting global vehicle data/information to Vehicle root branch using deprecation
* 87155cc  Gunnar Andersson  7 weeks ago (authored 7 weeks ago)  vss-tools: Bump submodule to include units update
* 153cfbf  Ryan Rowe  8 weeks ago (authored 3 months ago)  Add carets to clarify powers
* 488a45a  Ryan Rowe  8 weeks ago (authored 3 months ago)  Rename units pa, kpa to Pa, kPa
* b35f4f3  Daniel (dakrbo)  9 weeks ago (authored 3 months ago)  Added various combustion engine properties
* 1565e2f  Daniel Wilms  2 months ago (authored 2 months ago)  Updated vss-tools for instance support
* 9c1b89b  SebastianSchildt  3 months ago (authored 3 months ago)  Fix/fix units again (#212)
* 6127e87  Sebastian Schildt  3 months ago (authored 3 months ago)  Bump HUGO version in travis
* d82d16f  Sebastian Schildt  3 months ago (authored 3 months ago)  Update Hugo theme version to work with newer Hugo releases

I'm wondering about the fact that there is a commit named "Release 2.0" on the master branch, but not on the Release branch, and that there are many commits on master also not on the release branch.  Should they be there?

**2) What's the overall plan for maintenance branches for 2.0 and beyond?**

I'd recommend we keep a maintenance branch for each.  Should it be called `/release/2.0` like now?
Another naming convention I have used in the past is `2.x` for the maintenance branch, *if* we will reuse it for all minor releases after 2.0.   It might be a reasonable way to avoid a maintenance branch for every minor release (too many).

danielwilms, 2021-02-02:
> I'm wondering about the fact that there is a commit named "Release 2.0" on the master branch, but not on the Release branch,

There the documentation of the release has been added. For 1.0 we used the tag and I would keep it that way.

> Commits in origin/master but not in origin/release/2.0:

At some point we branched off, but I didn't remember, that we discussed, that the commits < ef683c4 should have been added. I should have created a ticket. Like this I forgot. that should be changed

> Another naming convention I have used in the past is 2.x for the maintenance branch, if we will reuse it for all minor releases after 2.0. It might be a reasonable way to avoid a maintenance branch for every minor release (too many).

That sounds reasonable. At the end the tag should count.

danielwilms, 2021-02-02:
@gunnarx  https://github.com/GENIVI/vehicle_signal_specification/blob/master/RELEASE_PROCESS.md

danielwilms, 2021-04-20:
fixed, if there's something open still, please reopen

closed 2021-04-20 by danielwilms