COVESA/vehicle_signal_specification #238: Discuss optionality of signals

gunnarx, 2021-02-11:
Moved from #137 

@erikbosch asked:
> Would the "layer" concept in some way be possible to use to specify which datapoints that are supported/mandatory/optional. Today we have a lot of datapoints, but no way to specify if the datapoints are mandatory or optional.

The layers concept is there to allow companies, or the "VSS-project", or another project, or anyone (who uses it is independent of the definition of VSS layers as a concept), to add _any_ additional metadata it might need.

What you describe could of course be an example given in the README, but I would like to avoid it because it suggests things that we are not agreed on regarding if and how to comply with the specification.  It is a separate question if VSS project, or CVII, _or any other project_, should be involved in specifying some things as mandatory and some as optional as part of a (requirement) specification.

There are many other alternatives than defining this.
 1)  Consider all signals as "mandatory"
 2) Allow companies to implement what they want and see where the industry takes this
 3)  Let either GENIVI or some totally different organization define _a separate activity_ to check compliance with the specification and give out approvals/certifications, and so on.  To be clear, at least GENIVI has no "compliance specification" tied to VSS at this point.  All of these things are questions that will not be concluded quickly, so let's discuss them separately from the mechanics of the layers.

gunnarx, 2021-02-11:
My brief opinion is I'm not actually that keen on discussing this at this time - I mean I think we have bigger fish to fry at the moment.  But I put it here to isolate the discussion from #137, if others are interested in discussing it now.

erikbosch, 2024-02-27:
I am doing some cleanup closing all issues created 2021 or earlier where there have been no activity 2023 or later. If you still consider this issue relevant feel free to reopen it add and add a comment on how you would like to see progress on this one.

closed 2024-02-27 by erikbosch