COVESA/vehicle_signal_specification #239: Field Naming convention

geojins, 2021-02-12:
Hi,
I noticed that few fields in the spec violates the naming convention from other.  For example.

Vehicle.IgnitionOn --> Field Starts with Upper case.  - This is the majority
Vehicle.accelerationTime --> Field starts with lower case. ( very few fields are like this)

Wondering if this is intended or an issue ?. 

Thanks,
Jins

erikbosch, 2021-02-25:
@gunnarx - do we have a style guideline somewhere? If not, would it be an idea to somewhere in a repo create one, to e.g. state that "CamelCase" shall be the preferred pattern. If we have an agreement on style, we can then decide if we want to change existing items or leave them as is. If we decide to change all then we need to align on how to do it - it can be a bit cumbersome if we want to add a new item "AccelerationTime" but keep "accelertionTime" as deprecated for some time.

danielwilms, 2021-03-30:
would be good to add to documentation. @danielwilms make proposal

PR #389 (2022-01-12): Adapting signal names to naming rules
Making sure first character is upper case and that boolean signals
starts with "Is" as described by naming conventions in https://covesa.github.io/vehicle_signal_specification/rule_set/basics/

Also some other minor changes, like adding period at end of description when missing and removing branch if branch only has one member in a few situations.

Solves the warnings reported by https://github.com/COVESA/vss-tools/pull/128

Fixes #384 
Fixes #293 
Fixes #239
closed 2022-01-12 by danielwilms