COVESA/vehicle_signal_specification #246: Meaning of "aggregate" for branches?

erikbosch, 2021-02-25:
In https://genivi.github.io/vehicle_signal_specification/rule_set/branches/ we describe aggregate. Well we do not describe it that well:

`Defines whether or not this branch is an aggregate. If not defined, this defaults to false.`

It is used in a few cases where data logically "belongs together". But do we know of any tooling actually relying on "aggregate" in any kind? Or is it just handled as informal information? I think it would be good to improve the documentation on "aggregate" and I can create a PR for that, but then I first need to understand the background for it, and make sure that we still consider it useful (otherwise we can deprecate it)

Example usage:
```
- CurrentLocation:
  type: branch
  aggregate: true
  description: The current latitude and longitude of the vehicle.
```

SebastianSchildt, 2021-03-03:
I would suggest to drop "aggregate" altogether. For things where it is obvious (i.e. location) and App/user will know it (and anyway it is not forbidden to track latitude only...), and complex constraints like  required time synchronicity across different sensors on different branch can not expressed with this method anyway (and again would be dependent on an application).

danielwilms, 2021-03-09:
@magnusfeuer any opinion on that one?

PR #254 (2021-04-06): Removing aggregate keyword
Fixes #246

After this has been merged,  a separate PR for vss-tools
shall be merged to give error if keyword "aggregate" is used.
commit 3e39406b (2021-04-06): Removing aggregate keyword

Fixes #246

Removing aggregate as keyword in actual VSS specification,
but leaving it as allowed syntax when defining a branch.
(I.e. vss-tools shall still accept usage of it)

The intention to leave it is as it might be reused in e.g.
the future layer concept, i.e. an instantiation of VSS might
specify that a particular branch shall be handled as an aggregate.
closed 2021-04-06 by danielwilms