COVESA/vehicle_signal_specification #247: Signals in Android VHAL

UlfBj, 2021-02-25:
Have we done an analysis/comparison with this set of 139 Android signals?
https://android.googlesource.com/platform/hardware/interfaces/+/master/automotive/vehicle/2.0/types.hal

erikbosch, 2021-02-25:
There is a draft/incomplete comparison in https://at.projects.genivi.org/jira/browse/AASIG-122

(note that there are some suffix/format issues with v1.2 in the issue, I could not open by double-clicking but opening it from within excel worked fine)

In addition to signals themselves, it is interesting to see that Android has selected as different approach for "area-handling" when you have multiple instances of something, by having bitmasks, like this example:

```
Example 2: A car has three seat rows with two seats in the front row (ROW_1_LEFT,
     *  ROW_1_RIGHT) and three seats in the second (ROW_2_LEFT, ROW_2_CENTER,
     *  ROW_2_RIGHT) and third rows (ROW_3_LEFT, ROW_3_CENTER, ROW_3_RIGHT). There
     *  are three temperature control units -- driver side, passenger side, and rear.
     *   - A reasonable way to map HVAC_TEMPERATURE_SET to AreaIDs is a three
     *     element array:
     *     - ROW_1_LEFT
     *     - ROW_1_RIGHT
     *     - ROW_2_LEFT | ROW_2_CENTER | ROW_2_RIGHT | ROW_3_LEFT | ROW_3_CENTER | ROW_3_RIGHT

```
This can be compared to the current VSS approach where we use row/pos as default, and it is not obvious how we would map the setup in the example above to a VSS deployment. 
```
- Station:
  type: branch
  instances:
    - Row[1,4]
    - ["Left","Right"]
  description: HVAC for single station in the vehicle
#include SingleHVACStation.vspec Station
```

danielwilms, 2021-03-02:
@gunnarx: Ongoing work in GENIVI, work planned and it's coming

closed 2024-02-27 by erikbosch